bohdan-s/SunGather #156: Reinstallation failure in Home Assistant add-on

nithikalisamy, 2023-12-12:
Hi, I had an issue with MQTT so uninstalled it.

However, upon trying to install, it fails everytome and couldnt install it anymore. I had tried rebooting the RPi4 and restarted the HA.
But nothing seems to be working to make it install again.

Please let me know if there is a fix or workaround.  Appreciate your inputs. Thanks.

**Error:**
Failed to install add-on
The command '/bin/ash -o pipefail -c pip install --no-cache-dir --upgrade pycryptodomex==3.11.0 -r requirements.txt' returned a non-zero code: 1

**HA log:**
Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:138
Integration: Home Assistant Supervisor ([documentation](https://www.home-assistant.io/integrations/hassio), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+hassio%22))
First occurred: 10:33:02 AM (14 occurrences)
Last logged: 11:28:36 PM

Failed to to call /addons/7b536ee6_sungather/install - The command '/bin/ash -o pipefail -c pip install --no-cache-dir --upgrade pycryptodomex==3.11.0 -r requirements.txt' returned a non-zero code: 1

nithikalisamy, 2023-12-12:
I have now tried to restore from the back up where i did not install SunGather also now reports same error.

I have tried to run the eblwo commands that was suggested for similar issue
ha su repair
ha core rebuild
ha host reboot

But none of this solved this issue yet.

kirangp, 2023-12-13:
I am also getting the same error when I tried reinstalling. Nothing has worked.

nithikalisamy, 2023-12-13:
> I am also getting the same error when I tried reinstalling. Nothing has worked.

Agree, Past two days tried to see all possible solution but then nothing worked so far.

kirangp, 2023-12-13:
@nithikalisamy See this msg in Discord. The developer has acknowledged the issue. 
https://discord.com/channels/936031869001158666/1012552777509118072/1184362917236322376

Unfortunately, until there is a fix, there is not much we can do.

nithikalisamy, 2023-12-13:
> @nithikalisamy See this msg in Discord. The developer has acknowledged the issue. https://discord.com/channels/936031869001158666/1012552777509118072/1184362917236322376
> 
> Unfortunately, until there is a fix, there is not much we can do.

Thanks for the link. Could you pease post here if there is any update.

bohdan-s, 2024-04-05:
@nithikalisamy this should now be resolved in v0.1.4 of the HA addon / v0.5.3 of SunGather.