bohdan-s/SunGather #157: Support for SH8.0RT-20

AlexanderLanin, 2023-12-21:
Can you add support for device type 3602 = SH8.0RT-20

Compare: https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/c42255c9f841b0cc275b5bcec7918687b5c6cc61/modbus_sungrow.yaml#L1970

Maybe it would even be possible to support all unknown inverters?

AlexanderLanin, 2023-12-22:
Quick n dirty fix... at least it doesn't abort directly at startup...

in `registers-sungrow.yaml` search-replace `"SH8.0RT",` with `"SH8.0RT","SH8.0RT-20",` and add:
```
          - response: 0xE12
            value: "SH8.0RT-20"
```
to `device_type_code`.

Octofinger, 2023-12-22:
> Quick n dirty fix... at least it doesn't abort directly at startup...
> 
> in `registers-sungrow.yaml` search-replace `"SH8.0RT",` with `"SH8.0RT","SH8.0RT-20",` and add:
> 
> ```
>           - response: 0xE12
>             value: "SH8.0RT-20"
> ```
> 
> to `device_type_code`.

Or even simpler, hard set your inverter model to SH8.0RT in ``config.yaml``. Same registers so it will work the same.