bohdan-s/SunGather #158: SG15.RT: no "work_state_1", kills exporting to influxdb

pohly, 2024-01-07:
I recently had an incident where a SunGrow SG15.RT entered "state: not initialized" (from its web interface) and no longer produced anything. Power cycling it fixed this. I now checked whether I can store that state in influxdb to set up some alerting solution. 

I found "work_state_1" and "work_state_2" in the registers file and added that:

```
  # Output data to InfluxDB
  - name: influxdb
    enabled: True                           # [Optional] Default is False
    ...
    measurements:                           # [Required] Registers to publish to bucket
      - point: "work_state_1"
        register: "work_state_1"
      - point: "work_state_2"
        register: "work_state_2"
      - point: "energy"
        register: daily_power_yields
```

This didn't work. Debug output shows that only "work_state_1" is present, but with `None` as value:
```
Jan 07 14:42:10 xbmc conmon[496320]: +----------------------------------------------------------------------+
Jan 07 14:42:10 xbmc conmon[496320]: | Address | Register                            | Value                |
Jan 07 14:42:10 xbmc conmon[496320]: +---------+-------------------------------------+----------------------+
Jan 07 14:42:10 xbmc conmon[496320]: | ----    | device_type_code                    | SG15RT               |
Jan 07 14:42:10 xbmc conmon[496320]: | vr001   | run_state                           | ON                   |
Jan 07 14:42:10 xbmc conmon[496320]: | vr003   | last_reset                          | 2024-1-7 14:42:09    |
...
Jan 07 14:42:10 xbmc conmon[496320]: | 5038    | work_state_1                        | None                 |
```

Then exporting to influxdb dies:
```
Jan 07 14:42:10 xbmc conmon[496320]: Traceback (most recent call last):
Jan 07 14:42:10 xbmc conmon[496320]:   File "/usr/src/sungather/./sungather.py", line 190, in <module>
Jan 07 14:42:10 xbmc conmon[496320]:     main()
Jan 07 14:42:10 xbmc conmon[496320]:   File "/usr/src/sungather/./sungather.py", line 158, in main
Jan 07 14:42:10 xbmc conmon[496320]:     export.publish(inverter)
Jan 07 14:42:10 xbmc conmon[496320]:   File "/usr/src/sungather/exports/influxdb.py", line 64, in publish
Jan 07 14:42:10 xbmc conmon[496320]:     value = inverter.getRegisterValue(register) if type(inverter.getRegisterValue(register)) is str else float(inverter.getRegisterValue(register))
Jan 07 14:42:10 xbmc conmon[496320]:                                                                                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jan 07 14:42:10 xbmc conmon[496320]: TypeError: float() argument must be a string or a real number, not 'NoneType'
Jan 07 14:42:11 xbmc podman[496331]: 2024-01-07 14:42:11.01818019 +0100 CET m=+0.046560191 container died e41c7fefdc2a2af0a8bd1dae72c13f7e80cef0ce02a083e94b77a3913ade42e1 (image=docker.io/bohdans/sungather:latest, name=sungrow-15)
```

I suppose the influxdb code should handle `None` more gracefully.

Not sure whether I really want ` work_state_1`, though. Perhaps `run_state` is what I am looking for.

pohly, 2024-01-07:
Looks like I need `work_state_1` after all. The inverter is down again, with:
```
vr001	run_state	ON 
5038	work_state_1	None 
5081	work_state_2	None 
```

pohly, 2024-01-08:
I power cycled the inverter. Now I get something other than `None`:
```
vr001	run_state	ON 
5038	work_state_1	Run 
5081	work_state_2	Stop 
```

So it looks like it's only the "uninitialized" work state that isn't handled. Fixing the handling of `None` in `influxdb.py:64` such that missing values get skipped should be enough for me: then I can set up alerting with "unknown state for xxx minutes" as trigger.

stahlsau, 2026-09-10:
years later but still helpful - thanks for your post!
Apparently had the same issue with the influx export, though the inverter was running fine, work_state was 'none' and though the sungathering stopped.
Disabled work_state in the config and voila.

closed 2026-10-24 by bohdan-s