bohdan-s/SunGather #160: Failed to install add-on SunGather

Korsberg, 2024-01-13:
When try ti install I got this error , any way to fix this ?

Failed to install add-on
The command '/bin/ash -o pipefail -c pip install --no-cache-dir --upgrade pycryptodomex==3.11.0 -r requirements.txt' returned a non-zero code: 1

hossamnagy, 2024-02-04:
I ve got same error, restarting home assistant didn’t help

davidsddh, 2024-02-07:
The command '/bin/ash -o pipefail -c pip install --no-cache-dir --upgrade pycryptodomex==3.11.0 -r requirements.txt' returned a non-zero code: 1

Is this a simple typo /bin/[b]ash?

bohdan-s, 2024-02-07:
No. The pycryptodomex package is only source. So the docker builds on build. But HA has disabled builds. I have a fix but it requires a bit of rework and I haven’t had time to publish it yet.

djp3k05, 2024-02-19:
we hope that you will get the time to release the fix. thank you.

Wildvirus, 2024-02-24:
At console i got an error: 
   running build_ext
      building 'Cryptodome.Hash._MD2' extension
      creating build/temp.linux-x86_64-cpython-311
      creating build/temp.linux-x86_64-cpython-311/src
      gcc -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DPYCRYPTO_LITTLE_ENDIAN -DSYS_BITS=64 -DLTC_NO_ASM -Isrc/ -I/usr/include/python3.11 -c src/MD2.c -o build/temp.linux-x86_64-cpython-311/src/MD2.o
      error: command 'gcc' failed: No such file or directory

Perhaps it helps..... hoping to a solution