bohdan-s/SunGather #163: Data reported on webserver not appearing to HA > 'Unkown'

chrisborell, 2024-02-11:
I'm encountering an issue where data is being successfully scraped from the inverter and displayed on the webserver but isn't being forwarded to Home Assistant via MQTT. They report being 'Unknown'

- Active import from Grid
- Daily export to grid
- Daily import from grid

[image]

Address
--

---- | device_type_code | SG10RT
vr001 | run_state | ON
vr003 | last_reset | 2024-2-12 8:49:25
vr006 | daily_export_to_grid | 0.006833333333333333 kWh
vr007 | daily_import_from_grid | 0.16488055555555547 kWh
5001 | nominal_active_power | 10.0 kW
5002 | output_type | 2P
5003 | daily_power_yields | 2.3 kWh
5004 | total_power_yields | 16825.3 kWh
5006 | total_running_time | 3977 h
5008 | internal_temperature | 42.1 °C
5009 | total_apparent_power | 3108 VA
5011 | mppt_1_voltage | 440.0 V
5012 | mppt_1_current | 2.4 A
5013 | mppt_2_voltage | 394.9 V
5014 | mppt_2_current | 3.5 A
5017 | total_dc_power | 3334 W
5019 | phase_a_voltage | 231.1 V
5020 | phase_b_voltage | 0.0 V
5021 | phase_c_voltage | 0.0 V
5022 | phase_a_current | 13.4 A
5023 | phase_b_current | 0.0 A
5024 | phase_c_current | 0.0 A
5031 | total_active_power | 3108 W
5033 | total_reactive_power | -4 Var
5035 | power_factor | 1.0
5036 | grid_frequency | 49.99 Hz
5038 | work_state_1 | Dispatch Run
5049 | nominal_reactive_power | 6.0 kVar
5071 | array_insulation_resistance | 1556 k-ohm
5077 | active_power_regulation_setpoint | 10000 W
5079 | reactive_power_regulation_setpoint | 0 Var
5081 | work_state_2 | Run
5083 | meter_power | -360 W
5091 | load_power | 2748 W
5113 | daily_running_time | 126 min
5128 | monthly_power_yields | 670.8 kWh
5146 | negative_voltage_to_the_ground | 0.0 V
5147 | bus_voltage | 465.5 V
5150 | pid_work_state | None
5151 | pid_alarm_code | None
5216 | export_power | -328 W
5218 | power_meter | 2737 W
7013 | string_1_current | 0.0 A
7014 | string_2_current | 0.0 A
7015 | string_3_current | 0.0 A
5006 | start_stop | Start
5007 | power_limitation_switch | Enable
5008 | power_limitation_setting | 100.0 %
5019 | power_factor_setting | 0.0
5035 | night_svg_switch | Disable
5036 | reactive_power_adjustment_mode | Off
5037 | reactive_power_percentage_setting | 0.0 %
5039 | power_limitation_adjustment | 10.0 kW
5040 | reactive_power_adjustment | 0.0 kVar
5041 | pid_recovery | Disable
vr002 | timestamp | 2024-2-12 9:05:05
vr004 | export_to_grid | 360 W
vr005 | import_from_grid | 0 W

<br class="Apple-interchange-newline">

swordfish6975, 2024-02-12:
@chrisborell  same issue this is what fixed it for me, ssh to your home assistant server.

`ha core update --version 2024.1.5`

Think something broke in core version 2024.1.6+ or perhaps there is now a incompatibility with SunGather and the new core versions.

swordfish6975, 2024-02-12:
Alternatively it could be something to do with this here.

[image]

[image]

perhaps these two need to be just total  (not total_increasing) now in the config files?

bohdan-s, 2024-02-12:
Yes. There have been some breaking changes I need to address. Sorry

swordfish6975, 2024-02-12:
All good, let me know if I can help any any way :)

swordfish6975, 2024-02-14:
I have updated back to the latest HA Core 2024.2.1 and commented out the following two lines from my config.yaml

`last_reset_value_template: "{{ value_json.last_reset }}"`  

(note: its there twice once for daily_import_from_grid and once for daily_export_to_grid)

After a reboot it seems to be working again, I  will monitor for adverse affects and report back in a few days.

weichenw, 2024-02-15:
I have changed the 'state_class' to 'total'. Seems to be running fine now. Currently on 2024.2.1

swordfish6975, 2024-02-18:
Removing the "last_reset_value_template" seems to have caused no adverse affects for me 👍 

@weichenw interested as to what the outcome was leaving it there and making it state_class:total?

weichenw, 2024-02-18:
> Removing the "last_reset_value_template" seems to have caused no adverse affects for me 👍
> 
> @weichenw interested as to what the outcome was leaving it there and making it state_class:total?

It seems to work for me as per normal.

skye-harris, 2024-02-20:
Much thanks to you both, @weichenw @swordfish6975

I have used the same solution here as provided by @swordfish6975 and this also seems to have worked without issue (touch wood) :)

patrikbloemer, 2024-03-07:
> I have updated back to the latest HA Core 2024.2.1 and commented out the following two lines from my config.yaml
>
> `last_reset_value_template: "{{ value_json.last_reset }}"`
>
> (note: its there twice, [...]

In what file do you edit? Where do I find it?
/Patrik

chrfritsch, 2024-03-07:
> In what file do you edit? Where do I find it?

It`s in `/root/share/SunGather` on my Raspberry