bohdan-s/SunGather #170: Mqtt Component communicating with Mqtt server

snaven1, 2024-03-11:
I have had Sungather working flawlessly for the last year without issue. I upgraded the base operating system from Fedora 36 to Fedora 39
This broke Sungather completely due to various library issues. I was forced to reinstall SunGather as per online guide. This went well without issue. Using the same configuration file, SunGather will not communicate to my MQTT server (different box), otherwise works correctly in terms of extracting data from the SunGrow inverter. This is the error I am receiving. 

Mar 11 10:36:06 hsmart.homebase python3[34892]: 2024-03-11 10:36:06 WARNING  MQTT: Server Error; Server not configured
Mar 11 10:36:06 hsmart.homebase python3[34892]: MQTT: Server Error; Server not configured
Mar 11 10:36:06 hsmart.homebase python3[34892]: 2024-03-11 10:36:06 INFO     Next scrape in 15 secs

On using mosquitto tools ($mosquitto_pub) from the same machine, this works correctly and talks to the mosquitto mqtt server. 
I have also included a snippet from my configuration file (config.yaml) related to the mqtt server part

  # Publish Registers to MQTT / Home Assistant
  - name: mqtt
    enabled: true                           # [Optional] Default is False
    host: 192.168.100.86                    # [Required] IP or Hostname of MQTT Server
    port: 1883                              # [Optional] Default 1883
    topic: "SunGrow/SH10RT/registers"      # [Optional] Default: "inverter/{model}/registers", Variable {model} will be replaced with model number of inverter
    username: "mqtt"                        # [Optional] Username is MQTT server requires it
    password: "mqtt"                        # [Optional] Password is MQTT server requires it
    client_id: "SunGather1"                 # [Optional] Client id for mqtt connection. Defaults to "SunGather", need to be set to different values if you want to run multiple instances of sungather.
    homeassistant: True                     # [Optional] Default False, Publish discovery and sensors for HomeAssistant
    ha_discovery: False

Thanks for any assistance

bohdan-s, 2024-03-11:
Start the script with debug logging and grab the first 20 odd lines. My guess is the model isn’t detecting right which is part of the MQTT topic.

snaven1, 2024-03-11:
After running script manually, with debugging .. the critical part of the log related to the mqtt issue is as follows ...

2024-03-11 11:31:25 INFO     Loading Export: exports\webserver
2024-03-11 11:31:25 INFO     Webserver: Configured
2024-03-11 11:31:25 INFO     Loading Export: exports\jsonserver
2024-03-11 11:31:25 INFO     JSON-Export: Configured
2024-03-11 11:31:25 INFO     Loading Export: exports\mqtt
2024-03-11 11:31:25 ERROR    Failed loading export: Unsupported callback API version: version 2.0 added a callback_api_version, see migrations.md for details
                             Please make sure mqtt.py exists in the exports folder
Failed loading export: Unsupported callback API version: version 2.0 added a callback_api_version, see migrations.md for details
                             Please make sure mqtt.py exists in the exports folder
2024-03-11 11:31:25 DEBUG    Checking Modbus Connection
2024-03-11 11:31:25 DEBUG    Modbus, Session is still connected
2024-03-11 11:31:25 DEBUG    Scraping: read, 4949:50
2024-03-11 11:31:25 DEBUG    load_registers: read, 4949:50
2024-03-11 11:31:25 DEBUG    Current transaction state - TRANSACTION_COMPLETE
20

The mqtt.py file is located correctly and with full read permissions available.

Thanks

snaven1, 2024-03-11:
Problem solved.  The Paho Python MQTT module needs to be downgraded from V2 to V1.6. Essentially, the V2 Api is incompatible with V1 scripts, hence the downgrade. The solutions are outlined at the following URL
https://stackoverflow.com/questions/77984857/paho-mqtt-unsupported-callback-api-version-error

Hope this helps

bullantmm, 2024-09-08:
I am having a similar issue with a new setup (haven't yet successfully connected to MQTT). I am running Mosquitto and Sungather in Docker containers but I get the following: 
`2024-09-08 23:56:45 INFO     Loading Export: exports\webserver
2024-09-08 23:56:45 INFO     Webserver: Configured
2024-09-08 23:56:45 INFO     Loading Export: exports\mqtt
2024-09-08 23:56:45 ERROR    Failed loading export: 'NoneType' object has no attribute 'replace'
			     Please make sure mqtt.py exists in the exports folder
Failed loading export: 'NoneType' object has no attribute 'replace'
			     Please make sure mqtt.py exists in the exports folder`

Any help would be appreciated