bohdan-s/SunGather #175: Possibility of support for additional battery devices and their modbus registers?

paulhomes, 2024-04-06:
Creating this issue as per suggestion from @bohdan-s in Discord conversation https://discord.com/channels/936031869001158666/1148822763780460564/1225730715090423858 to capture the Home Assistant modbus yaml config fragment I am using to get more information from my SBR battery stacks (including module level data such as per-module min/max cell voltages).

[sungrow_modbus_battery_fragment.yaml.txt](https://github.com/bohdan-s/SunGather/files/14893257/sungrow_modbus_battery_fragment.yaml.txt)

I originally found this yaml in the discussion at https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/discussions/133 and made a few tweaks to get it working with my 2 battery stacks.

Does Sungather support getting modbus data from other devices on other slave/device ids? I use Sungather on my SH10RT inverter on slave/device id 1, as well as Home Assistant direct modbus reading device id 1 for the inverter, 200 for the first battery stack and 201 for the second battery stack. I'll keep using direct modbus with HA for now but I imagine other Sungather users with SBR battery stacks might find it useful if their battery device data was also available via Sungather.

I seem to remember I needed to upgrade the battery firmware from SBRBCU-S_22011.01.11 to SBRBCU-S_22011.01.19 before I was able to access the battery modbus registers on device id 200. I also use the dedicated LAN port on my SH10RT for HA modbus reading. I am not sure if the battery registers are available via the WiNet-S dongle.

vetinari, 2024-05-12:
> I am not sure if the battery registers are available via the WiNet-S dongle.

At least I can read the battery serial and firmware (SBRBCU-S_22011.01.17) and some more values through an SH5.0RT, device id 2 via the dongle...

vetinari, 2024-05-13:
```
battery_temperature => 15.700000 °C
total_battery_discharge => 154.700000 kWh
battery_serial => S230221XXXX
battery_firmware => SBRBCU-S_22011.01.17
battery_voltage => 329.600000 V
battery_current => 0.400000 A
battery_soc => 39.100000 %
total_battery_charge => 227.200000 kWh
battery_soh => 100.000000 %
```

seems to work, missing values (compared to your HA config) are all 0

kalaws, 2024-05-22:
Did you somehow integrate this with Sungather, or by forking? Or using mkaiser's integration in addition?

vetinari, 2024-05-22:
@kalaws If you're talking me: I have my own private version in go (a rewrite of  sungather).

paulhomes, 2024-05-22:
@kalaws In my case, whilst I started with sungather and still have an instance running, these days I mostly use a slightly modified mkaiser HA modbus config primarily for battery charge/discharge control based on wholesale electricity prices. I also look at the individual battery module and stack data because I am troubleshooting reduced capacity due to unbalanced stacks and unbalanced modules within a stack which I am trying to resolve with Sungrow.

paulhomes, 2024-05-22:
The battery module level data allows me to see voltage charts in HA like this max one when the stacks get full:

[image]

and this min one when the stacks are empty:

[image]