bohdan-s/SunGather #184: Futurerequest: add a wallbox e.g. AC011E-01

HubertusH, 2024-05-27:
Hello everyone,
would it also be possible to query a wallbox such as the AC011E-01? Then you could also read out this data. This would be practical as it would also allow you to calculate the charging power. Currently, the inverter only displays the total power consumed.

Thank you and best regards
Hubertus

michbeck100, 2024-05-27:
No, Sungather is designed for Sungrow inverters. Not for wallboxes

HubertusH, 2024-05-29:
oh dear ;-) ok i have changed that to:
```
scan: # these have to be 1 less than the first register
  - read:
    - start: 21199
      range: 10

  - hold:
    - start: 21199
      range: 10
```

but the error remains - I may not have understood the "hold" parameter.

Thanks & greetings
Hubertus

ludifu, 2024-05-29:
Change this:

```
        - response: 0x01
          value: "1ph"

scan: # these have to be 1 less than the first register
```

into this:

```
        - response: 0x01
          value: "1ph"
  - scan:
 
scan: # these have to be 1 less than the first register
```

HubertusH, 2024-05-29:
Thank you - one step further!
Because the error message with the "year" came up, I set use_local_time to true. But it still doesn't run smoothly. 
The key "year" is not set in my config.
```
python3 sungather.py -c /home/pi/SunGather/Wallbox/config.yaml -r /home/pi/SunGather/Wallbox/registers-wallbox.yaml -v 30 --runonce
2024-05-29 10:19:43 INFO     Starting SunGather 0.5.2
2024-05-29 10:19:43 INFO     Need Help? https://github.com/bohdan-s/SunGather
2024-05-29 10:19:43 INFO     NEW HomeAssistant Add-on: https://github.com/bohdan-s/hassio-repository
2024-05-29 10:19:44 INFO     Loaded config: /home/pi/SunGather/Wallbox/config.yaml
2024-05-29 10:19:44 INFO     Loaded registers: /home/pi/SunGather/Wallbox/registers-wallbox.yaml
2024-05-29 10:19:44 INFO     Registers file version: 0.2.3
+----------------------------------------------+
| Inverter Configuration Settings              |
+----------------------------------------------+
| Config             | Value                   |
+--------------------+-------------------------+
| host               | 192.168.179.9           |
| port               | 502                     |
| timeout            | 10                      |
| retries            | 3                       |
| RetryOnEmpty       | False                   |
| model              | AC011E-01               |
| serial_number      | None                    |
| level              | 2                       |
| scan_interval      | 30                      |
| use_local_time     | True                    |
| smart_meter        | False                   |
| connection         | modbus                  |
| slave              | 2                       |
| start_time         |                         |
+----------------------------------------------+
2024-05-29 10:19:47 ERROR    Failed to scrape: 'year'
Traceback (most recent call last):
  File "/home/pi/SunGather/Wallbox/sungather.py", line 156, in main
    success = inverter.scrape()
  File "/home/pi/.local/lib/python3.9/site-packages/SungrowClient/SungrowClient.py", line 379, in scrape
    del self.latest_scrape["year"]
KeyError: 'year'
Failed to scrape: 'year'
Traceback (most recent call last):
  File "/home/pi/SunGather/Wallbox/sungather.py", line 156, in main
    success = inverter.scrape()
  File "/home/pi/.local/lib/python3.9/site-packages/SungrowClient/SungrowClient.py", line 379, in scrape
    del self.latest_scrape["year"]
KeyError: 'year'
2024-05-29 10:19:47 WARNING  Data collection failed, skipped exporting data. Retying in 30 secs
Data collection failed, skipped exporting data. Retying in 30 secs

```
Thanks & greetings
Hubertus

ludifu, 2024-05-29:
It really only scrapes what you defined in ```registers:```. If the yaml you posted earlier is complete there are only 3 registers defined, not including year.

Background information: The ```scan:```section is used to first read all data in the defined address areas. In a second step Sungather extracts all defined registers from these areas. After this it has some logic to convert date information into a date field and delete the single year, month, day, etc. fields. Thus the ```del self.latest_scrape["year"]```code snippet. This cannot be changed by configuration. So you must pull all those date fields.

HubertusH, 2024-05-29:
Thank you !

i defined "Year":
```
    - name: "Year"
      level: 2
      address: 21224
      datatype: "U16"
      unit: "YYYY"

```
but "year" is case sensitiv and now with "year" lowercase it works.
So - for you: a query of the wallbox connected to the inverter (in this case the AC011E-01) also works.

MANY THANKS for the support. Great.

Sunny greetings
Hubertus

ludifu, 2024-05-29:
Well, thanks to you for researching this wallbox stuff!

HubertusH, 2024-05-29:
Super
only a few formatting problems remain ;-)
The documentary writes:
```
"_address	name				description									unit	type		len	factor	offset	formula	role	room	poll	wp	cw	isScale"
"21200		Total_Energy		Bisher geladene Gesamtleistung				wh		uint32be	2	1			state		true	true	false	false"
"21202		SetOutI				Ausgangsstrom | 0,1A | 6A-63A				A		uint16be	1	1			state		true	true	false	false"
"21203		PhaseSwitch			PhaseSwitch | 0=3ph 1=1ph							uint16be	1	1			state		true	true	false	false"
"21210		Charger_Enable		Charger Enable |  0=disable 1=enable				uint16be	1	1			state		true	true	false	false"
"21211		START_STOP_CHARGING	Ladung Starten | Stoppen 0=Start 1=Stop				uint16be	1	1			state		true	true	false	false"
"21212		Total_Energy		Total_Energy								Wh		uint32be	2	1			state		true	true	false	false"
"21224		Year				Year												uint16be	1	1			state		true	true	false	false"
"21225		Month				Month												uint16be	1	1			state		true	true	false	false"
"21226		Day					Day													uint16be	1	1			state		true	true	false	false"
"21227		Hour				Hour												uint16be	1	1			state		true	true	false	false"
"21228		Minute				Minute												uint16be	1	1			state		true	true	false	false"
"21229		Second				Second												uint16be	1	1			state		true	true	false	false"
"21230		Dev_Address			Bus Adresse | deafult 248							uint16be	1	1			state		true	true	false	false"
"21231		Enery_per_Km		Energie pro Km										uint16be	1	1			state		true	true	false	false"
```

the results from modbus tool fit so far:
```

modbus 192.168.179.9 -s 2 21200 = 38174 // according to APP "38.2" would be correct
modbus 192.168.179.9 -s 2 21202 = 60 
modbus 192.168.179.9 -s 2 21203 = 0        // only "0" or "1"
modbus 192.168.179.9 -s 2 21210 = 1        // only "0" or "1" 
modbus 192.168.179.9 -s 2 21211 = 0        // only "0" or "1"
modbus 192.168.179.9 -s 2 21230 = 248    //  default is 248 !
modbus 192.168.179.9 -s 2 21231 = 50

```

but the output from Sungather doesn't really fit:
```
+----------------------------------------------------------------------+
| Address | Register                            | Value                |
+---------+-------------------------------------+----------------------+
| ----    | device_type_code                    | AC011E-01            |
| 21200   | Total_Energy                        | 1093795840 wh        |
| 21202   | SetOutI                             | 842543673 A          |
| 21203   | PhaseSwitch                         | 12856                |
| 21210   | ChargerEnable                       | 0                    |
| 21211   | StartStopCharging                   | 0                    |
| 21212   | TotalEnergy2                        | 0 wh                 |
| 21230   | Dev_Adress                          | 17709                |
| 21231   | Energy_per_Km                       | 12853                |
| vr002   | timestamp                           | 2024-05-29 13:36:02  |
+----------------------------------------------------------------------+
```

the matching register configuration:
```
version:  0.2.3
vendor: Sungrow
registers:
  - read:
    - name: "Total_Energy"
      level: 2
      address: 21200
      datatype: "U32"
      unit: "wh"
    - name: "SetOutI"
      level: 2
      address: 21202
      datatype: "U32"
      unit: "A"
    - name: "PhaseSwitch"
      level: 2
      address: 21203
      datatype: "U16"
#      datarange:
#        - response: 0x00
##          value: "3ph"
#        - response: 0x01
#          value: "1ph"
    - name: "ChargerEnable"
      level: 2
      address: 21210
      datatype: "U16"
    - name: "StartStopCharging"
      level: 2
      address: 21211
      datatype: "U16"
    - name: "TotalEnergy2"
      level: 2
      address: 21212
      datatype: "U32"
      unit: "wh"
    - name: "year"
      level: 2
      address: 21224
      datatype: "U16"
      unit: "YYYY"
    - name: "month"
      level: 2
      address: 21225
      datatype: "U16"
      unit: "MM"
    - name: "day"
      level: 2
      address: 21226
      datatype: "U16"
    - name: "hour"
      level: 2
      address: 21227
      datatype: "U16"
    - name: "minute"
      level: 2
      address: 21228
      datatype: "U16"
    - name: "second"
      level: 2
      address: 21229
      datatype: "U16"
    - name: "Dev_Adress"
      level: 2
      address: 21230
      datatype: "U16"
    - name: "Energy_per_Km"
      level: 2
      address: 21231
      datatype: "U16"

```

maybe you have an idea how to get the values displayed correctly

Otherwise, YES, it was great to work this out together.

Greetings
Hubertus

HubertusH, 2024-05-29:
_Another important note for other users: querying the registers of the wallbox only works on the WInetS connection - these registers are not available on the LAN connection!_

HubertusH, 2024-05-31:
Now the chaos is complete ;-)

I have looked in your "registers-sungrow". There the registers for "year" etc. are in the hold section. If I do the same I get the error message: 
```
2024-05-31 10:38:49 ERROR    Failed to scrape: 'year'
Traceback (most recent call last):
  File "/home/pi/SunGather/Wallbox/sungather.py", line 156, in main
    success = inverter.scrape()
  File "/home/pi/.local/lib/python3.9/site-packages/SungrowClient/SungrowClient.py", line 379, in scrape
    del self.latest_scrape["year"]
KeyError: 'year'
Failed to scrape: 'year'
Traceback (most recent call last):
  File "/home/pi/SunGather/Wallbox/sungather.py", line 156, in main
    success = inverter.scrape()
  File "/home/pi/.local/lib/python3.9/site-packages/SungrowClient/SungrowClient.py", line 379, in scrape
    del self.latest_scrape["year"]
KeyError: 'year'

```
I then put these registers back in the read section - at least that part works.

An additional question: my serial number of the wallbox is "A2292803764" but with the query data type "UTF-8" I only get "A2292803".

I will now test the registers one by one and get back to you with the next problem :-))

Greetings
Hubertus

Translated with www.DeepL.com/Translator (free version)

ludifu, 2024-05-31:
The registers for year etc. are not the same as those from the inverter, because it is another unit / slave.

Testing register by register is a good idea. (However year and month etc. will probably not be in different sections.) But those registers that deliver strange values are good candidates. The data you reported is not only off by an accuracy factor ...

The problem with serial number and UTF8 is a bug in Sungather. (There has been an issue about this here before: The person reporting this had the problem of two identical serial numbers because they were different in only the last digit.) Sungather simply assumes too few bytes of data. And this is hardcoded way in the original version of Sungather and cannot be configured.

Btw. are you sure you get "A2292803" and not "A229280376" (only omitting the "4")?

HubertusH, 2024-06-21:
wow - the biggest mistake on my part was that an offset of "1" must be observed for the INet connection. But not for all registers. :-)

Now, after a lot of testing, it works:

```
+----------------------------------------------------------------------+
| Address | Register                            | Value                |
+---------+-------------------------------------+----------------------+
| ----    | device_type_code                    | AC011E-01            |
| vr001   | run_state                           | OFF                  |
| vr003   | last_reset                          | 2024-06-21 10:10:25  |
| vr006   | daily_export_to_grid                | 0.0 kWh              |
| vr007   | daily_import_from_grid              | 0.0 kWh              |
| 21201   | serial_number                       | A229280376           |
| 21207   | PhaseSwitch                         | 0                    |
| 21262   | Rated_Voltage                       | 230 V                |
| 21263   | WorkMode                            | Netzwerk             |
| 21268   | Charger_Status                      | Enable               |
| 21300   | Total_Energy                        | 172.7 kWh            |
| 21302   | Charging_Voltage_R                  | 0.0 V                |
| 21303   | Charging_Current_R                  | 0.0 A                |
| 21304   | Charging_Voltage_S                  | 0.0 V                |
| 21305   | Charging_Current_S                  | 0.0 A                |
| 21306   | Charging_Voltage_T                  | 0.0 V                |
| 21307   | Charging_Current_T                  | 0.0 A                |
| 21308   | Charging_Power                      | 0.0 kW               |
| 21310   | Charging_Engery                     | 12.67 kWh            |
| 21314   | Start_Mode                          | None                 |
| 21316   | Power_Regulation                    | forbidden            |
| 21317   | Charging_Status                     | Idle                 |
| 21318   | Charge_Start_Time                   | 1718883655           |
| 21320   | Charge_End__Time                    | 1718894336           |
| 21316   | Hold-Status                         | 0                    |
| vr002   | timestamp                           | 2024-06-21 10:10:25  |
| vr004   | export_to_grid                      | 0 W                  |
| vr005   | import_from_grid                    | 0 W                  |
+----------------------------------------------------------------------+
Logged 28 registers to Console

```
I have also attached my register file
[registers-wallbox.yaml.txt](https://github.com/user-attachments/files/15925114/registers-wallbox.yaml.txt)

The "PhaseSwitch" register has not yet been confirmed. But you can also see the function, whether single-phase or multi-phase, from the Charging_xxx_y registers. If only one phase is charging, the power is Charging_Voltage_R * Charging_Current_R for all three phases (simplified formula) Charging_Voltage_R * Charging_Current_R * 1.732 (square root of 3).

Small problems remain:

- Sereinumber should be a string with a length of 20 - but only 10 of 11 characters are returned.
Original: A2292803764 A229280376 is returned. If you shift the register by one, the "6" appears but then the "A" at the beginning is missing.
This is not dramatic, but is it possible to read out a byte sequence and then assemble it yourself?
- Is there a built-in formula to convert the Unix timestamps to Date/Time?

Sungrow has confirmed that the registers of the wallbox are only available via the WINetS connection. If you could use two instances of sungather this would not be a problem.

Greetings
Hubertus

HubertusH, 2025-02-17:
Hi,
I have updated to the current version 0.5.2 and now get the error message with the same config and register file: 

```
2025-02-17 10:47:05 ERROR    Failed to scrape: time data '8410-3-16707 12337:12619:11585' does not match format '%Y-%m-%d %H:%M:%S'
Traceback (most recent call last):
  File "/home/pi/SunGather/SunGather/sungather.py", line 159, in main
    success = inverter.scrape()
              ^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/SungrowClient/SungrowClient.py", line 460, in scrape
    elif datetime.strptime(self.latest_scrape['last_reset'], date_format).date() < datetime.strptime(self.latest_scrape['timestamp'], date_format).date():
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/_strptime.py", line 568, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/_strptime.py", line 349, in _strptime
    raise ValueError("time data %r does not match format %r" %
ValueError: time data '8410-3-16707 12337:12619:11585' does not match format '%Y-%m-%d %H:%M:%S'
Failed to scrape: time data '8410-3-16707 12337:12619:11585' does not match format '%Y-%m-%d %H:%M:%S'
Traceback (most recent call last):
  File "/home/pi/SunGather/SunGather/sungather.py", line 159, in main
    success = inverter.scrape()
              ^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/SungrowClient/SungrowClient.py", line 460, in scrape
    elif datetime.strptime(self.latest_scrape['last_reset'], date_format).date() < datetime.strptime(self.latest_scrape['timestamp'], date_format).date():
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/_strptime.py", line 568, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/_strptime.py", line 349, in _strptime
    raise ValueError("time data %r does not match format %r" %
ValueError: time data '8410-3-16707 12337:12619:11585' does not match format '%Y-%m-%d %H:%M:%S'
2025-02-17 10:47:05 WARNING  Data collection failed, skipped exporting data. Retying in 30 secs
Data collection failed, skipped exporting data. Retying in 30 secs

```
It is about the registers vr002 and vr003
`| vr002   | timestamp                           | 8410-3-16707 12337:12619:11585  |
`
`| vr003   | last_reset                          | 8410-3-16707 12337:12619:11585  |`

Everything is ok with the query of the Sungrow inverter - only refers to the query of the wallbox developed here

any idea how I can find/fix the error?

Thank you and best regards
Hubertus

HubertusH, 2025-02-17:
Yes, I did it - it works. 
The “solution” lies in the definition of the registers. SunGrow has apparently corrected this in the wallbox. Previously you had to define the registers “YEAR” etc. in the -read section, which worked even though they are -hold registers. Now this no longer works (checked with modbus-tcp, the registers can only be queried as “hold”). If you now move the registers from the -read section to -hold, they are not recognized and the error message appears again that “year” is not defined.
I can live with “local time” - of course it would be better if the registers were also recognized in the -hold section.

Best regards and thanks
Hubertus

Translated with www.DeepL.com/Translator (free version)