bohdan-s/SunGather #185: SG10RS - Configured to use total_active_power but not configured to scrape this register

jkerekes99, 2024-06-04:
Hi, apologies if this has been covered before, however, I haven't been able to find the answer searching.

My inverter is:
SG10RS with WiNet-S Dongle

I am using the Home Assistant Addon via the repo.
I am getting all the entities except for:
* Active Generation
* Active Export to Grid
* Active Import from Grid

The log shows the following:
2024-06-04 12:22:38 INFO     Loaded config: /share/SunGather/config.yaml
2024-06-04 12:22:39 INFO     Loaded registers: registers-sungrow.yaml
2024-06-04 12:22:39 INFO     Registers file version: 0.2.4
2024-06-04 12:22:42 ERROR    MQTT: Configured to use total_active_power but not configured to scrape this register
MQTT: Configured to use total_active_power but not configured to scrape this register

Not sure what I'm doing wrong, there is not much to configure in the addin and as I said the other entities are working well.

bohdan-s, 2024-06-04:
Smart _meter needs true and level set to 2

jkerekes99, 2024-06-04:
> Smart _meter needs true and level set to 2

Thanks for the quick reply.
I have this set already, I still get that error in the log and those entities are missing:
[image]

I have a S100 smart meter.

jkerekes99, 2024-06-04:
So I opened /share/SunGather/config.yaml and it is showing level: 1.
So it is NOT taking the "level" changes made in the addon configuration tab.

hadimiqbal, 2024-06-19:
Any update on this , i have the same inverter but cannot get HA to show anything

jkerekes99, 2024-06-19:
It is working for me, just not:

- Active Generation
- Active Export to Grid
- Active Import from Grid

[image]

hadimiqbal, 2024-06-19:
having said that, it just started when i moved the port back to 502
[image]

jkerekes99, 2024-06-19:
@hadimiqbal 
So you get entities I don't.
But your Inverter is a different model to mine.

My inverter is a SG10RS
Yours is a SG10RT.

hadimiqbal, 2024-06-19:
Yeah its actually RS i was just trying different name if it made a difference

jkerekes99, 2024-06-19:
> Yeah its actually RS i was just trying different name if it made a difference

Interesting, where did you change the model?
You are getting "Active Power" where I am not.

hadimiqbal, 2024-06-19:
i did it in the UI but the YAML looks like this for now
[image]

hadimiqbal, 2024-06-19:
its funny i am not seeing the ones you see, do you think it may take time ?

hadimiqbal, 2024-06-19:
would you know how toi generate more than these 3
[image]

jkerekes99, 2024-06-19:
Not sure, I don't think so.
My entities appeared immediately.

Are you running the latest firmware on your inverter?
[image]

[image]

jkerekes99, 2024-06-19:
> would you know how toi generate more than these 3

They just appeared after entering the IP address into the GUI config and hitting start.

jkerekes99, 2024-06-23:
Ok, so I managed to get all the supported entities working:
[image]

I had to however set:
level:3
in the config.yaml (via terminal, not via HA adding).

Setting to 2 does not work.

However the issue I am getting now is the log file is filled with errors trying to scrape the other unsupported entities:
`Traceback (most recent call last):
  File "/venv/lib/python3.12/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/venv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/venv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2024-06-23 17:47:56 WARNING  Modbus connection failed
Modbus connection failed
2024-06-23 17:47:58 ERROR    index out of range
index out of range
2024-06-23 17:47:58 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/venv/lib/python3.12/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/venv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/venv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/venv/lib/python3.12/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/venv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/venv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request`

Is there a way to be able to pick the entities/registers you want the addon to scrape? rather than just choosing a level?
Ideally, I would love level 2 to work on the SG10RS inverters, however it doesn't.

mitch-1211, 2024-07-15:
Hi @jkerekes99, I am also using this add on with the SG10RS. I'm able to get the following sensors with level 3:

```
Sungrow SG10RS Active Export to Grid
sensor.sungrow_sg5rs_active_export_to_grid
MQTT
—
—

Sungrow SG10RS Active Generation
sensor.sungrow_sg5rs_active_generation
MQTT
—
—

Sungrow SG10RS Active Import from Grid
sensor.sungrow_sg5rs_active_import_from_grid
MQTT
—
—

Sungrow SG10RS Daily Export to Grid
sensor.sungrow_sg10rs_daily_export_to_grid
MQTT
—
—

Sungrow SG10RS Daily Generation
sensor.sungrow_sg10rs_daily_generation
MQTT
—
—

Sungrow SG10RS Daily Import from Grid
sensor.sungrow_sg10rs_daily_import_from_grid
MQTT
—
—

Sungrow SG10RS Last reset
sensor.sungrow_sg10rs_last_reset
MQTT
—
—

Sungrow SG10RS Load Power
sensor.sungrow_sg10rs_load_power
MQTT
—
—

Sungrow SG10RS Meter Power
sensor.sungrow_sg10rs_meter_power
MQTT
—
—

Sungrow SG10RS Power State
binary_sensor.sungrow_sg10rs_power_state
MQTT
—
—

Sungrow SG10RS Temperature
sensor.sungrow_sg10rs_temperature
MQTT
—
—

Sungrow SG10RS Total Export Energy
sensor.sungrow_sg5rs_total_export_energy
MQTT
—
—

Sungrow SG10RS Total Import Energy
sensor.sungrow_sg5rs_total_import_energy
MQTT
—
—

Sungrow SG10RS Total Power Yields
sensor.sungrow_sg5rs_total_power_yields
MQTT
```

No errors in log like you have. Can you share your config.yaml? (located under /share/SunGather/config.yaml)