bohdan-s/SunGather #186: Problems publishing collected data via MQTT

hwinkelmann, 2024-06-08:
Hi,

I am struggling with the export of the harvested data via MQTT to home assistant. The data collection works nicely, in the web UI, i get a nice list of address registers and values, and the data is looking good, including the serial number. However, pushing that data to mosquitto fails.

I could post the entire debug log, but a whole lot deals with the state machine collecting the modbus data, which is obviously working fine. So I've reverted to the `info` log level, if that's not enough i'm happy to provide the debug log.

```
/sungather.py:139: SyntaxWarning: invalid escape sequence '\{'
  logging.info(f"Loading Export: exports\{export.get('name')}")
2024-06-08 11:33:04 INFO     Starting SunGather 0.5.2
2024-06-08 11:33:04 INFO     Need Help? https://github.com/bohdan-s/SunGather
2024-06-08 11:33:04 INFO     NEW HomeAssistant Add-on: https://github.com/bohdan-s/hassio-repository
2024-06-08 11:33:04 INFO     Loaded config: /share/SunGather/config.yaml
2024-06-08 11:33:05 INFO     Loaded registers: registers-sungrow.yaml
2024-06-08 11:33:05 INFO     Registers file version: 0.2.4
2024-06-08 11:33:05 INFO     Logging to console set to: INFO
2024-06-08 11:33:05 INFO     Loading SungrowClient 0.1.0
2024-06-08 11:33:05 INFO     Modbus client is not connected, attempting to reconnect
2024-06-08 11:33:05 INFO     Connection: ModbusTcpClient(10.0.0.158:502)
2024-06-08 11:33:08 WARNING  Modbus connection failed
Modbus connection failed
2024-06-08 11:33:08 INFO     Model detection failed, please set model in config.py
2024-06-08 11:33:09 WARNING  ModbusTcpClient(10.0.0.158:502): Connection unexpectedly closed 1.480055 seconds into unbounded read after returning 29 bytes
ModbusTcpClient(10.0.0.158:502): Connection unexpectedly closed 1.480055 seconds into unbounded read after returning 29 bytes
2024-06-08 11:33:09 INFO     Detected Serial: A22A050910
2024-06-08 11:33:09 INFO     Closing Session: ModbusTcpClient(10.0.0.158:502)
2024-06-08 11:33:09 INFO     Loading Export: exports\webserver
2024-06-08 11:33:09 INFO     Webserver: Configured
2024-06-08 11:33:09 INFO     Loading Export: exports\mqtt
2024-06-08 11:33:09 ERROR    Failed loading export: 'NoneType' object has no attribute 'replace'
			     Please make sure mqtt.py exists in the exports folder
Failed loading export: 'NoneType' object has no attribute 'replace'
			     Please make sure mqtt.py exists in the exports folder
2024-06-08 11:33:09 INFO     Modbus, Connecting new session
2024-06-08 11:33:09 INFO     Setting Initial Daily registers; daily_export_to_grid, daily_import_from_grid, last_reset
2024-06-08 11:33:09 INFO     Inverter: Successfully scraped in 0.8014 secs
2024-06-08 11:33:09 WARNING  MQTT: Server Error; Server not configured
MQTT: Server Error; Server not configured
2024-06-08 11:33:09 INFO     Closing Session: ModbusTcpClient(10.0.0.158:502)
2024-06-08 11:33:09 INFO     Next scrape in 29 secs
```

I am no python dev, but the first line looks like a syntax error?

The addon's config looks like this:
```yaml
host: 10.0.0.158
timeout: 10
scan_interval: 30
connection: modbus
smart_meter: false
custom_mqtt_server: false
log_console: INFO
model: SH6.0RT
```

If you have any questions, please let me know! And thanks for your work!

michbeck100, 2024-06-08:
I’ve seen this error mostly when mosquitto doesn’t run before sungather starts or gets restarted. Sungather misses some resilience here.

hwinkelmann, 2024-06-08:
I've disabled sungather and restarted the system and home assistant. Once mosquitto was up, I've started sungarher again. But the log looks still as above, so I assume it must be something else 😕

hwinkelmann, 2024-06-08:
Misclicked, sorry

LokAndFeel, 2024-08-29:
For me it helps to downgrade paho-mqtt to version 1.6. There is a breaking change in version 2.0

baylf2000, 2024-09-05:
> For me it helps to downgrade paho-mqtt to version 1.6. There is a breaking change in version 2.0

Could you explain a little about how you did this please?

bullantmm, 2024-09-08:
@baylf2000 I am having the same issue. @LokAndFeel how can you do this? I am current running Mosquitto in a Docker container with no issues with other containers.

bohdan-s, 2024-09-08:
Hi

This issue is the model is not detecting correctly, and the model makes up part of the MQTT topic. 
Setting the model manually in the config will fix this. 
New version will fix this.

bullantmm, 2024-09-09:
Thanks so much @bohdan-s! I am now getting some data from my SG8.0RS but log shows the following errors:

`2024-09-09 10:35:51 ERROR    index out of range
2024-09-09 10:35:51 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2024-09-09 10:35:51 WARNING  Modbus connection failed
2024-09-09 10:35:51 ERROR    index out of range
2024-09-09 10:35:51 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2024-09-09 10:35:51 WARNING  Modbus connection failed
2024-09-09 10:35:51 ERROR    index out of range
2024-09-09 10:35:51 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2024-09-09 10:35:51 WARNING  Modbus connection failed
2024-09-09 10:35:51 ERROR    index out of range
2024-09-09 10:35:51 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2024-09-09 10:35:51 WARNING  Modbus connection failed
2024-09-09 10:35:51 ERROR    index out of range
2024-09-09 10:35:51 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2024-09-09 10:35:51 WARNING  Modbus connection failed
2024-09-09 10:35:51 ERROR    index out of range
2024-09-09 10:35:51 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2024-09-09 10:35:51 WARNING  Modbus connection failed
2024-09-09 10:35:52 ERROR    index out of range
2024-09-09 10:35:52 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
[... cut ...]
    self.client.framer.processIncomingPacket(response,
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/virtualenv/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2024-09-09 10:35:53 WARNING  Modbus connection failed
`