bohdan-s/SunGather #187: Connection Failed - Not all arguments converted during string formatting

rmcastelao, 2024-06-12:
Hi,

I'm able to connect over HTTP but can't fetch actual data. Seems like an unhandled exception when parsing:

```
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.12/logging/__init__.py", line 1160, in emit
    msg = self.format(record)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/logging/__init__.py", line 999, in format
    return fmt.format(record)
           ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/logging/__init__.py", line 703, in format
    record.message = record.getMessage()
                     ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/logging/__init__.py", line 392, in getMessage
    msg = msg % self.args
          ~~~~^~~~~~~~~~~
TypeError: not all arguments converted during string formatting
Call stack:
  File "/sungather.py", line 195, in <module>
    main()
  File "/sungather.py", line 125, in main
    if not inverter.checkConnection():
  File "/venv/lib/python3.12/site-packages/SungrowClient/SungrowClient.py", line 90, in checkConnection
    return self.connect()
  File "/venv/lib/python3.12/site-packages/SungrowClient/SungrowClient.py", line 73, in connect
    try: self.client.connect()
  File "/venv/lib/python3.12/site-packages/SungrowModbusWebClient/SungrowModbusWebClient.py", line 101, in connect
    logging.warning("Connection Failed", payload_dict['result_msg'] )
Message: 'Connection Failed'
Arguments: ('I18N_COMMON_INTER_ABNORMAL',)
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.12/logging/__init__.py", line 1160, in emit
    msg = self.format(record)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/logging/__init__.py", line 999, in format
    return fmt.format(record)
           ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/logging/__init__.py", line 703, in format
    record.message = record.getMessage()
                     ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/logging/__init__.py", line 392, in getMessage
    msg = msg % self.args
          ~~~~^~~~~~~~~~~
TypeError: not all arguments converted during string formatting
Call stack:
  File "/sungather.py", line 195, in <module>
    main()
  File "/sungather.py", line 125, in main
    if not inverter.checkConnection():
  File "/venv/lib/python3.12/site-packages/SungrowClient/SungrowClient.py", line 90, in checkConnection
    return self.connect()
  File "/venv/lib/python3.12/site-packages/SungrowClient/SungrowClient.py", line 73, in connect
    try: self.client.connect()
  File "/venv/lib/python3.12/site-packages/SungrowModbusWebClient/SungrowModbusWebClient.py", line 101, in connect
    logging.warning("Connection Failed", payload_dict['result_msg'] )
Message: 'Connection Failed'
Arguments: ('I18N_COMMON_INTER_ABNORMAL',)
2024-06-12 12:14:47 ERROR    Error: Connection to inverter failed: 192.168.1.199:502
Error: Connection to inverter failed: 192.168.1.199:502
Error: Connection to inverter failed: 192.168.1.199:502
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

wellsy57, 2024-07-02:
Hi there,

I am also having possibly the same problem myself?

After nearly 12 months running seamlessly and no changes to config now I have this in my log:

`s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
/sungather.py:139: SyntaxWarning: invalid escape sequence '\{'
  logging.info(f"Loading Export: exports\{export.get('name')}")
2024-07-02 11:21:51 INFO     Starting SunGather 0.5.2
2024-07-02 11:21:51 INFO     Need Help? https://github.com/bohdan-s/SunGather
2024-07-02 11:21:51 INFO     NEW HomeAssistant Add-on: https://github.com/bohdan-s/hassio-repository
2024-07-02 11:21:51 INFO     Loaded config: /share/SunGather/config.yaml
2024-07-02 11:21:52 INFO     Loaded registers: registers-sungrow.yaml
2024-07-02 11:21:52 INFO     Registers file version: 0.2.4
2024-07-02 11:21:52 INFO     Logging to console set to: DEBUG
2024-07-02 11:21:52 DEBUG    Inverter Config Loaded: {'host': '192.168.1.243', 'port': 8080, 'timeout': 10, 'retries': 5, 'slave': 1, 'scan_interval': 30, 'connection': 'http', 'model': 'SH6.0RS', 'smart_meter': False, 'use_local_time': False, 'log_console': 'DEBUG', 'log_file': 'OFF', 'level': 2}
2024-07-02 11:21:52 INFO     Loading SungrowClient 0.1.0
2024-07-02 11:21:52 DEBUG    Checking Modbus Connection
2024-07-02 11:21:52 INFO     Modbus client is not connected, attempting to reconnect
2024-07-02 11:21:52 INFO     Connection: SungrowModbusWebClient_0.3.2(192.168.1.243:8082)
2024-07-02 11:21:52 DEBUG    Connection to websocket server established: ws://192.168.1.243:8082/ws/home/overview
2024-07-02 11:21:52 DEBUG    {'result_code': 1, 'result_msg': 'success', 'result_data': {'service': 'connect', 'token': '9fa19a17-34b7-4f53-ae48-05bc649e8962', 'uid': 1, 'tips_disable': 0, 'virgin_flag': 0, 'isFirstLogin': 0, 'forceModifyPasswd': 0}}
2024-07-02 11:21:52 INFO     Token Retrieved: 9fa19a17-34b7-4f53-ae48-05bc649e8962
2024-07-02 11:21:52 DEBUG    Requesting Device Information
2024-07-02 11:21:52 DEBUG    {'result_code': 401, 'result_msg': 'I18N_COMMON_INTER_ABNORMAL', 'result_data': {'service': 'devicelist'}}
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.12/logging/__init__.py", line 1160, in emit
    msg = self.format(record)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/logging/__init__.py", line 999, in format
    return fmt.format(record)
           ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/logging/__init__.py", line 703, in format
    record.message = record.getMessage()
                     ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/logging/__init__.py", line 392, in getMessage
    msg = msg % self.args
          ~~~~^~~~~~~~~~~
TypeError: not all arguments converted during string formatting
Call stack:
  File "/sungather.py", line 195, in <module>
    main()
  File "/sungather.py", line 125, in main
    if not inverter.checkConnection():
  File "/venv/lib/python3.12/site-packages/SungrowClient/SungrowClient.py", line 90, in checkConnection
    return self.connect()
  File "/venv/lib/python3.12/site-packages/SungrowClient/SungrowClient.py", line 73, in connect
    try: self.client.connect()
  File "/venv/lib/python3.12/site-packages/SungrowModbusWebClient/SungrowModbusWebClient.py", line 101, in connect
    logging.warning("Connection Failed", payload_dict['result_msg'] )
Message: 'Connection Failed'
Arguments: ('I18N_COMMON_INTER_ABNORMAL',)
--- Logging error ---
Traceback (most recent call last):
[... 18 lines cut ...]
  File "/venv/lib/python3.12/site-packages/SungrowClient/SungrowClient.py", line 90, in checkConnection
    return self.connect()
  File "/venv/lib/python3.12/site-packages/SungrowClient/SungrowClient.py", line 73, in connect
    try: self.client.connect()
  File "/venv/lib/python3.12/site-packages/SungrowModbusWebClient/SungrowModbusWebClient.py", line 101, in connect
    logging.warning("Connection Failed", payload_dict['result_msg'] )
Message: 'Connection Failed'
Arguments: ('I18N_COMMON_INTER_ABNORMAL',)
2024-07-02 11:21:52 ERROR    Error: Connection to inverter failed: 192.168.1.243:8080
Error: Connection to inverter failed: 192.168.1.243:8080
Error: Connection to inverter failed: 192.168.1.243:8080
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

Wondering if a recent HA update is the problem?

Thanks in advance for any clues that may help

Cheers Steve

Cuupa, 2024-08-12:
I have the same problem:

Environment: 
SunGather is running in a docker container:

docker-compose of the sungather part:
```yaml
sungather:
    image: bohdans/sungather
    restart: unless-stopped
    ports:
      - '18087:8080'
    volumes:
      - /volume2/docker/PRODUCTION/energy-monitoring/sungather/config/config.yaml:/config/config.yaml
      - /volume2/docker/PRODUCTION/energy-monitoring/sungather/logs:/logs
    environment:
      TZ: "Europe/Berlin"
      PUID: 1060
      PGID: 100
```

This is my config.yaml:

```yaml
inverter:
  host: 10.80.23.13                        # [Required] IP Address of the Inverter or Dongle
  port: 8082                               # [Optional] Default for modbus is 502, for http is 8082
  # timeout: 10                             # [Optional] Default is 10, how long to wait for a connection
  # retries: 3                              # [Optional] Default is 3, how many times to retry if connection fails
  # slave: 0x01                             # [Optional] Default is 0x01
  # scan_interval: 30                       # [Optional] Default is 30
  connection: http                     # [Required] options: modbus, sungrow, http
  model: "SH8.0RT"                        # [Optional] This is autodetected on startup, only needed if detection issues or for testing
                                            # See model list here: https://github.com/bohdan-s/SunGather#supported
  # serial: xxxxxxxxxx                      # [Optional] This is autodetected on startup, only needed if detection issues or for testing, used as a unique ID
  # smart_meter: True                       # [Optional] Default is False, Set to true if inverter supports reading grind / house consumption
  # use_local_time: False                   # [Optional] Default False, Uses Inventer time, if true it uses PC time when updating timestamps (e.g. PVOutput)
  # log_console: INFO                       # [Optional] Default is WARNING, Options: DEBUG, INFO, WARNING, ERROR
  # log_file: DEBUG                         # [Optional] Default is OFF, Options: OFF, DEBUG, INFO, WARNING, ERROR
  # level: 1                                # [Optional] Set the amount of information to gather
                                            # 0 = Model and Solar Generation, 
                                            # 1 (default) = Useful data, all required for exports, 
                                            # 2 everything your Inverter supports, 
                                            # 3 Everything from every register 

# If you do not want to use a export, you can either remove the whole configuration block
# or set enabled: False
exports:
  # Print Registers to console, good for debugging / troubleshooting
  - name: console         
    enabled: True                          # [Optional] Default is False

  # Runs a simple Webserver showing Config and last read registers
  # Access at http://localhost:8080 or http://[serverip]:8080
  - name: webserver 
    enabled: True                           # [Optional] Default is False
    # port: 8080                            # [Optional] Default is 8080

  # Output data to InfluxDB
  - name: influxdb
    enabled: True                          # [Optional] Default is False
    url: "http://10.80.22.5:18086"            # [Optional] Default URL is http://localhost:8086
    token: "<REDACTED>"                            # [Required] API Token OR Username:Password (for influxdb v1.8x comment it out and use username + password)
    # username:                             # [Optional] Username if not using token
    # password:                             # [Optional] Password if not using token
    org: "Default"                          # [Required] InfluxDB Organization (for influxdb v1.8x this will be ignored)
    bucket: "SunGather"                     # [Required] InfluxDB Bucket (for influxdb v1.8x this is the database name)
    measurements:                           # [Required] Registers to publish to bucket
      - point: "power"
        register: daily_power_yields
[... 129 lines cut ...]
2024-08-12T15:56:28.813818489Z   File "/usr/local/lib/python3.12/logging/__init__.py", line 703, in format
2024-08-12T15:56:28.813903087Z     record.message = record.getMessage()
2024-08-12T15:56:28.813964567Z                      ^^^^^^^^^^^^^^^^^^^
2024-08-12T15:56:28.814006225Z   File "/usr/local/lib/python3.12/logging/__init__.py", line 392, in getMessage
2024-08-12T15:56:28.814075348Z     msg = msg % self.args
2024-08-12T15:56:28.814134343Z           ~~~~^~~~~~~~~~~
2024-08-12T15:56:28.814190012Z TypeError: not all arguments converted during string formatting
2024-08-12T15:56:28.814239608Z Call stack:
2024-08-12T15:56:28.814359485Z   File "/opt/sungather/sungather.py", line 195, in <module>
2024-08-12T15:56:28.814443382Z     main()
2024-08-12T15:56:28.814501202Z   File "/opt/sungather/sungather.py", line 125, in main
2024-08-12T15:56:28.814549140Z     if not inverter.checkConnection():
2024-08-12T15:56:28.814608345Z   File "/opt/virtualenv/lib/python3.12/site-packages/SungrowClient/SungrowClient.py", line 90, in checkConnection
2024-08-12T15:56:28.814691939Z     return self.connect()
2024-08-12T15:56:28.814749709Z   File "/opt/virtualenv/lib/python3.12/site-packages/SungrowClient/SungrowClient.py", line 73, in connect
2024-08-12T15:56:28.814813219Z     try: self.client.connect()
[... cut ...]
2024-08-12T15:56:28.814922185Z     logging.warning("Connection Failed", payload_dict['result_msg'] )
2024-08-12T15:56:28.814984111Z Message: 'Connection Failed'
2024-08-12T15:56:28.815039932Z Arguments: ('I18N_COMMON_INTER_ABNORMAL',)
2024-08-12T15:56:28.815084293Z 2024-08-12 17:56:28 ERROR    Error: Connection to inverter failed: 10.80.23.13:8082
2024-08-12T15:56:28.815178788Z Error: Connection to inverter failed: 10.80.23.13:8082
2024-08-12T15:56:28.815302274Z Error: Connection to inverter failed: 10.80.23.13:8082

```

Cuupa, 2024-08-12:
I've debugged the current main branch against my inverter.
I don't get the `TypeError` on this branch with my `config.yaml` (I just took the example and changed the IP address), but I also don't get any valid response from my inverter:

I have a breakpoint at `SungrowModbusWebClient.py` in the `connect`-function.
A connection to the inverter can be established and I get the following response (I've removed the token, because I don't know if it's security related, but it looks like a UUID to me):
```
{
	"result_code":	1,
	"result_msg":	"success",
	"result_data":	{
		"service":	"connect",
		"token":	"<redacted>",
		"uid":	1,
		"tips_disable":	1,
		"virgin_flag":	0,
		"isFirstLogin":	0,
		"forceModifyPasswd":	0
	}
}
```

The token is then successfully extracted and it's calling the inverter a second time (see line 87 in `SungrowModbusWebClient.py`)

```python
logging.debug("Requesting Device Information")
self.ws_socket.send(json.dumps({ "lang": "en_us", "token": self.ws_token, "service": "devicelist", "type": "0","is_check_token": "0" }))
#self.ws_socket.send(json.dumps({ "lang": "en_us", "token": self.ws_token, "service": "runtime" }))
result =  self.ws_socket.recv()
```
This is where it crashes. When I execute the request, I get the following response from my smart inverter:
```JSON
{
"result_code":	401,
"result_msg":	"I18N_COMMON_INTER_ABNORMAL",
"result_data":	{
    "service":	"devicelist"
  }
}
```

Another strange thing I observed: 
My inverter got updated to the latest version, which seems to be "M_WiNet-S_V01_V01_A" and since then I can only use the WebUI of the inverter for 30 seconds max. After a short amount of time, I'll get logged out with the following message:

"Die Anzahl der Zugriffsberechtigungen ist uberschritten, es wurde abgemeldet. Bitte bei Modbus Pausezeiten und/oder IP-Proxy"
 (yes, the sentence ends unexpectedly) 
[image]

monojk, 2024-10-17:
I have the same problem original after a firmware update of my SunGrow system today. 
````
--- Logging error ---
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/logging/__init__.py", line 1083, in emit
    msg = self.format(record)
  File "/usr/local/lib/python3.9/logging/__init__.py", line 927, in format
    return fmt.format(record)
  File "/usr/local/lib/python3.9/logging/__init__.py", line 663, in format
    record.message = record.getMessage()
  File "/usr/local/lib/python3.9/logging/__init__.py", line 367, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/home/pi/SunGather/SunGather/sungather.py", line 198, in <module>
    main()
  File "/home/pi/SunGather/SunGather/sungather.py", line 124, in main
    if not inverter.checkConnection():
  File "/home/pi/.local/lib/python3.9/site-packages/SungrowClient/SungrowClient.py", line 90, in checkConnection
    return self.connect()
  File "/home/pi/.local/lib/python3.9/site-packages/SungrowClient/SungrowClient.py", line 73, in connect
    try: self.client.connect()
  File "/home/pi/.local/lib/python3.9/site-packages/SungrowModbusWebClient/SungrowModbusWebClient.py", line 101, in connect
    logging.warning("Connection Failed", payload_dict['result_msg'] )
Message: 'Connection Failed'
Arguments: ('I18N_COMMON_INTER_ABNORMAL',)

````

monojk, 2024-10-17:
Please see for my local fix of the problem: https://github.com/bohdan-s/SungrowModbusWebClient/issues/5#issuecomment-2420127866