bohdan-s/SunGather #188: Location of config.yaml for Home Assistant add-on

fullerms, 2024-06-17:
Hi all,

I have installed the sungather add-on  on Home Assistant running on a RPI4. The add-on seems to work fine, except the mqtt part. The logs show communication with the inverter, and values are being updated in the side bar. However, I don't see any entities being created.

This is the repository I used to install the add-on

`https://github.com/bohdan-s/hassio-repository`

**These are the errors I see in the sungather logs:**

`Skipping auto MQTT set up, please ensure MQTT settings are configured in /share/SunGather/config.yaml`
`WARNING  MQTT: Server Error; Server not configured. MQTT: Server Error; Server not configured`

I have tried adding mqtt details in the sungather configuration GUI as well as the yaml. I have created config.yaml in the following locations without any success
```
/SunGather
/share
/share/SunGather
```

This is my mqtt configuration in the config.yaml files
```
 # Publish Registers to MQTT / Home Assistant
  - name: mqtt
    enabled: True                          # [Optional] Default is False
    host: 192.168.x.xx                     # [Required] IP or Hostname of MQTT Server 
    # port: 1883                           # [Optional] Default 1883
    # topic: "SunGather/{serial_number}"   # [Optional] Default: "SunGather/{serial_number}"
    username: **                    # [Optional] Username is MQTT server requires it
    password: **               # [Optional] Password is MQTT server requires it
    client_id: **                  # [Optional] Client id for mqtt connection. Defaults to Serial Number.
    homeassistant: True
```

FWIW, I have Z2M configured on the same HA server and entities are being created for zigbee devices. So no issues with the mqtt setup AFAIK.

Can someone point me to the location where the config.yaml file is supposed to be placed? Or is something else missing?

bohdan-s, 2024-06-17:
There is a bug with the model detection on newer models. 
try setting the model to SG8.0RT and see if that works.

fullerms, 2024-06-17:
This is my inverter configuration

inverter:
  host: 192.168.x.x                   # [Required] IP Address of the Inverter or Dongle
  # port: 502                               # [Optional] Default for modbus is 502, for http is 8082
  # timeout: 10                             # [Optional] Default is 10, how long to wait for a connection
  # retries: 3                              # [Optional] Default is 3, how many times to retry if connection fails
  # slave: 0x01                             # [Optional] Default is 0x01
  # scan_interval: 30                       # [Optional] Default is 30
  connection: modbus                        # [Required] options: modbus, sungrow, http
  model: "SG8.0RT"                          # [Optional] This is autodetected on startup, only needed if detection issues or for testing

jkerekes99, 2024-06-23:
> This is my inverter configuration. I guess I must start with the location of the config.yaml file.

There should already be a file there you can edit.
The location should be:
/share/SunGather/config.yaml

fullerms, 2024-06-26:
@jkerekes99  I have installed HA supervisor on a RPI4, and these are the version details.

Core 2024.6.4 
Supervisor 2024.06.2
Operating System 12.4
Frontend 20240610.1

I am unable to find the /share directory in the file system within HA. Do I need to ssh into the RPI4 and check?

Interestingly, I was able to install the [ModbusTCP2MQTT](https://github.com/MatterVN/ModbusTCP2MQTT) add-on without any issues and pull the data.

jkerekes99, 2024-06-27:
@fullerms install the terminal add-on:
[image]

fullerms, 2024-06-27:
@jkerekes99 thanks for the additional info. I eventually figured it out after posting my message. It will be nice if @bohdan-s can mention this somewhere in the documentation to help us noobs :)

I managed to get everything running, but the Winnet-S2 dongle supplied with our SG8.0RT inverter keeps crashing after a few minutes. Ethernet Port is up, but no comms and the plant shows offline on the iSolarcloud app. I have to remove the dongle from the inverter and put it back to get it working again.

I have requested the installer to replace the dongle, but curious to know if others are facing the same issue?

bohdan-s, 2024-06-27:
Sorry. I have lots of notes of things to update ;)

Reduce your scan time. I find faster than 30s causes crashes on my winet-s (original)

jkerekes99, 2024-06-27:
> I managed to get everything running, but the Winnet-S2 dongle supplied with our SG8.0RT inverter keeps crashing after a few minutes. Ethernet Port is up, but no comms and the plant shows offline on [...]

Interesting you mention this.
My WiNet-S dongle is also down and now it wont come back online even after plugging back in (even after disabling SunGather), I can ping it and its connected to wifi, however wont connect to iSolarCloud (my solar installer is coming out to swap it).

fullerms, 2024-06-27:
> Sorry. I have lots of notes of things to update ;)
> 
> Reduce your scan time. I find faster than 30s causes crashes on my winet-s (original)

@bohdan-s I have gradually increased it to 120 seconds, and the dongle still keeps crashing :(

fullerms, 2024-06-27:
> Interesting you mention this. My WiNet-S dongle is also down and now it wont come back online even after plugging back in (even after disabling SunGather), I can ping it and its connected to wifi, [...]

I have connected to Ethernet :(

jkerekes99, 2024-06-27:
> I have gradually increased it to 120 seconds, and the dongle still keeps crashing :(

I think when you query registers it doesn't support, it crashes.
Hence why I asked if we could select the registers we want to scan.

fullerms, 2024-06-27:
> > I have gradually increased it to 120 seconds, and the dongle still keeps crashing :(
> 
> I think when you query registers it doesn't support, it crashes. Hence why I asked if we could select the registers we want to scan.

I have Sungather configured for level 3, that's the only way I am able to get smart meter information. And getting a lot of zeroes, and what you mentioned kinda makes sense.

@bohdan-s  do we have the option of choosing which register we want to read?

jkerekes99, 2024-06-27:
> I have Sungather configured for level 3

Same, that's the only way I get active generation.

fullerms, 2024-06-27:
I have been poking around the dongle and inverter settings a bit, and have come across two new options which I didn't know existed.

1. There is a configuration option in the WiNet-S2 dongle to forward modbus data to upto 3 IPs. But the fields are greyed out and I don't see an option to edit them.
2. FTP upload is also an option and the inverter uploads an xml file with some info.

I have attached the xml file (converted to txt) and a screen shot, perhaps worth exploring?

[image]
[24944_20240627_1415_1420.txt](https://github.com/user-attachments/files/16008027/24944_20240627_1415_1420.txt)

fullerms, 2024-06-27:
@bohdan-s   I see a  [registers-sungrow.yaml](https://github.com/bohdan-s/SunGather/blob/main/SunGather/registers-sungrow.yaml) file in the git repository.

1. Can we (users) remove the registers not supported by our inverters?
2. I don't see the file under /share/sungather, where will it be located in a hassio installation?

fullerms, 2024-06-28:
Did a little more digging in the registers.yaml file and found that meter power stats are enabled only for Hydrid inverters in level 2 scanning.

```
  - name: "meter_total_power"
      level: 2
      address: 5601
      datatype: "S32"
      unit: "W"
      models: ["SG8.0RT","SH5.0RS","SH3.6RS","SH4.6RS","SH6.0RS","SH10RT","SH10RT-V112","SH8.0RT","SH6.0RT","SH5.0RT","SH5.0RT-V112"]
    - name: "meter_phase_a_power"
      level: 2
      address: 5603
      datatype: "S32"
      unit: "W"
      models: ["SG8.0RT","SH5.0RS","SH3.6RS","SH4.6RS","SH6.0RS","SH10RT","SH10RT-V112","SH8.0RT","SH6.0RT","SH5.0RT","SH5.0RT-V112"]
    - name: "meter_phase_b_power"
      level: 2
      address: 5605
      datatype: "S32"
      unit: "W"
      models: ["SG8.0RT","SH5.0RS","SH3.6RS","SH4.6RS","SH6.0RS","SH10RT","SH10RT-V112","SH8.0RT","SH6.0RT","SH5.0RT","SH5.0RT-V112"]
    - name: "meter_phase_c_power"
      level: 2
      address: 5607
      datatype: "S32"
      unit: "W"
      models: ["SG8.0RT","SH5.0RS","SH3.6RS","SH4.6RS","SH6.0RS","SH10RT","SH10RT-V112","SH8.0RT","SH6.0RT","SH5.0RT","SH5.0RT-V112"]
```

@bohdan-s , I can confirm these registers work for the SG8.0RT model, Will appreciate some guidance on how to enable these registers for the SG8.0RT at Level 2 in a config file

bohdan-s, 2024-06-28:
So there are 2 checks. 
Level and models. As long as the level and model are there it should read it. 
Try setting the model in your config to a different model in that list and see if they load. 
the model was only used to filter the registers, next version will remove that for a more robust method.

jkerekes99, 2024-06-28:
> next version will remove that for a more robust method

I am so happy to hear this.
Thanks for all your hard work @bohdan-s

fullerms, 2024-06-28:
> So there are 2 checks. Level and models. As long as the level and model are there it should read it. Try setting the model in your config to a different model in that list and see if they load. the model was only used to filter the registers, next version will remove that for a more robust method.

I tried replacing the model number SG8.0RT with SH8.0RT;, but I am getting a lot of errors, and the add-on crashes.  I even tried adding those registers to the config.yaml file, but the addon wouldn't start.

Is there any way I can modify registers-sungrow.yaml in my installation, while you are working on the next version?

jkerekes99, 2024-06-28:
I finally got a working configuration.
I have an SG10RS inverter and I hard set the model to SG10RT and set the Level to 2.

Now I don't get any crashes or errors in the console and all the entities work.

fullerms, 2024-07-08:
@bohdan-s

I updated the registers-sungrow.yaml file and created a pull request to help with my configuration while you are working on the next update. First time ever using github, will appreciate if you can review and include the updated file.

https://github.com/fullerms/meterfix-sg-series-sungather-hassio-repository/pull/1

fullerms, 2024-07-18:
@bohdan-s could you please review and merge this pull request when you have a few spare moments? Including the Smart Meter data will help a lot.

https://github.com/bohdan-s/SunGather/pull/190