bohdan-s/SunGather #209: missing Registers publishing for inverter SH8.0RT

Surkel, 2024-11-15:
Hi,

today I tried to implement sungather for my SH8.0RT inverter with battery.
connection works probably, but I don't get published the registers.

<head></head>
Address
--

---- | device_type_code | SH8.0RT
vr001 | run_state | ON
vr003 | last_reset | 2024-11-15 15:14:10
vr006 | daily_export_to_grid | 0.0 kWh
vr007 | daily_import_from_grid | 0.03298333333333334 kWh
4950 | protocol_number | 1094791472
4952 | protocol_version | 16778752
4954 | arm_software_version | 16722
4969 | dsp_software_version | 19780
5001 | nominal_active_power | 8.0 kW
5002 | output_type | 3P4L
5003 | daily_power_yields | 2.6 kWh
5004 | total_power_yields | 205549 kWh
5008 | internal_temperature | 36.9 °C
5011 | mppt_1_voltage | 382.7 V
5012 | mppt_1_current | 0.0 A
5013 | mppt_2_voltage | 385.6 V
5014 | mppt_2_current | 0.0 A
5017 | total_dc_power | 0 W
5019 | phase_a_voltage | 230.7 V
5020 | phase_b_voltage | 230.2 V
5021 | phase_c_voltage | 232.8 V
5033 | total_reactive_power | -3 Var
5035 | power_factor | 1.0
5036 | grid_frequency | 499.8 Hz
5113 | daily_running_time | 0 min
5601 | meter_total_power | 440 W
5603 | meter_phase_a_power | 318 W
5605 | meter_phase_b_power | 34 W
5607 | meter_phase_c_power | 88 W
5622 | export_limit_min | 3400 W
5623 | export_limit_max | 12370 W
5628 | bdc_rated_power | 8000 W
5635 | bms_max_charging_current | 30 A
5636 | bms_max_discharging_current | 30 A
5723 | backup_phase_a_power | 0 W
5724 | backup_phase_b_power | 0 W
5725 | backup_phase_c_power | -2 W
5726 | total_backup_power | -2 W
13000 | system_state | Unknown
13001 | running_state | 40
13001 | state_power_generated_from_pv | 0
13001 | state_battery_charging | 0
13001 | state_battery_discharging | 0
13001 | state_load_active | 1
13001 | state_feed_into_grid | 0
13001 | state_import_from_grid | 1
13001 | state_power_generated_from_load | 0
13002 | daily_pv_generation | 2.9 kWh
13003 | total_pv_generation | 21347.9 kWh
13005 | daily_pv_export | 0.0 kWh
13006 | total_pv_export | 10834.5 kWh
13008 | load_power_hybrid | 402 W
13010 | export_power_hybrid | -444 W
13012 | daily_battery_charge_from_pv | 0.7 kWh
13013 | total_battery_charge_from_pv | 4817.4 kWh
13015 | co2_reduction | 0.0 kg
13017 | daily_direct_energy_consumption | 2.2 kWh
13018 | total_direct_energy_consumption | 5696.0 kWh
13020 | battery_voltage | 255.2 V
13021 | battery_current | 0.0 A
13022 | battery_power | 0 W
13023 | battery_level | 0.0 %
13024 | battery_state_of_healthy | 98.0 %
13025 | battery_temperature | 24.0 °C
13026 | daily_battery_discharge_energy | 0.4 kWh
13027 | total_battery_discharge_energy | 4538.2 kWh
13029 | self_consumption_of_day | 75.8 %
13030 | grid_state | None
13031 | phase_a_current | 0.4 A
13032 | phase_b_current | 0.4 A
13033 | phase_c_current | 0.4 A
13034 | total_active_power | -42 W
13036 | daily_import_energy | 13.4 kWh
13037 | total_import_energy | 12925.0 kWh
13040 | daily_charge_energy | 0.7 kWh
13041 | total_charge_energy | 4881.0 kWh
13043 | drm_state | 0
13045 | daily_export_energy | 0.0 kWh
13046 | total_export_energy | 10867.9 kWh
5006 | start_stop | Start
5007 | power_limitation_switch | Enable
5008 | power_limitation_setting | 100.0 %
5019 | power_factor_setting | 0.0
5036 | reactive_power_adjustment_mode | Off
5037 | reactive_power_percentage_setting | 0.0 %
5039 | power_limitation_adjustment | 0.0 kW
5040 | reactive_power_adjustment | 0.0 kVar
13050 | ems_mode_selection | Self-consumption mode
13051 | charge_discharge_command | Stop
13052 | charge_discharge_power | 0 W
13084 | start_charging_power | 0 W
13085 | start_discharging_power | 0 W
13086 | energy_meter_comm | Enabled
13087 | export_power_limitation | Disabled
13100 | soc_reserve | 0 %
33047 | battery_max_charge_power | 6000 W
33048 | battery_max_discharge_power | 6000 W
vr002 | timestamp | 2024-11-15 15:18:10
vr004 | export_to_grid | 0 W
vr005 | import_from_grid | 444 W

Configuration	Value
host	192.168.10.94
port	502
timeout	10
retries	3
RetryOnEmpty	False
model	SH8.0RT
serial_number	A215241031
level	2
scan_interval	30
use_local_time	False
smart_meter	False
connection	modbus
slave	1
start_time

[image]
but only 8 registers where published.

can anyone help?

closed 2024-11-16 by Surkel

JoJoJo2211, 2024-11-20:
Hello, that's the same for me. I think you can add them separately in the config since there are more than 100 entries. I thought there was a template? how did you solve that?

Surkel, 2024-11-20:
Yes, the First installed config file has only some sample Registers.
You have to enter the other Registers seperatly by your Self in the Config file.

Surkel, 2024-11-20:
no, now I use HA Modbus Intergation, there I don't need any changes at the config https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant