bohdan-s/SunGather #210: The option `last_reset_value_template` cannot be used together with state class `total_increasing`' when processing MQTT

Ducatel, 2024-12-01:
Hi

I'm trying to push my data to mqtt server which is used by Home assistant.

In HA, I face theses errors:

```
Error 'The option `last_reset_value_template` cannot be used together with state class `total_increasing`' when processing MQTT discovery message topic: 'homeassistant/sensor/A228055402_daily_export_to_grid/config', message: '{'state_topic': 'SunGather/A228055402', 'value_template': '{{ value_json.daily_export_to_grid | round(2) }}', 'device_class': 'energy', 'icon': 'mdi:transmission-tower-export', 'last_reset_value_template': '{{ value_json.last_reset }}', 'state_class': 'total_increasing', 'unique_id': 'sungather_daily_export_to_grid_A228055402', 'unit_of_measurement': 'kWh', 'device': {'manufacturer': 'Sungrow', 'model': 'SG60RS', 'identifiers': 'A228055402', 'via_device': 'SunGather', 'connections': [['address', '192.168.1.8']], 'name': 'Sungrow SG60RS'}, 'name': 'Daily Export to Grid'}'
Error 'The option `last_reset_value_template` cannot be used together with state class `total_increasing`' when processing MQTT discovery message topic: 'homeassistant/sensor/A228055402_daily_import_from_grid/config', message: '{'state_topic': 'SunGather/A228055402', 'value_template': '{{ value_json.daily_import_from_grid | round(2) }}', 'device_class': 'energy', 'icon': 'mdi:transmission-tower-import', 'last_reset_value_template': '{{ value_json.last_reset }}', 'state_class': 'total_increasing', 'unique_id': 'sungather_daily_import_from_grid_A228055402', 'unit_of_measurement': 'kWh', 'device': {'manufacturer': 'Sungrow', 'model': 'SG60RS', 'identifiers': 'A228055402', 'via_device': 'SunGather', 'connections': [['address', '192.168.1.8']], 'name': 'Sungrow SG60RS'}, 'name': 'Daily Import from Grid'}'
````

seems coming from the mqtt ha_sensor config.
I took it from the config sample:

````
- name: "Daily Export to Grid"
        sensor_type: sensor
        register: daily_export_to_grid
        device_class: energy
        state_class: total_increasing
        icon: "mdi:transmission-tower-export"
        value_template: "{{ value_json.daily_export_to_grid | round(2) }}"
        last_reset_value_template: "{{ value_json.last_reset }}"
      - name: "Daily Import from Grid"
        sensor_type: sensor
        register: daily_import_from_grid
        device_class: energy
        state_class: total_increasing
        icon: "mdi:transmission-tower-import"
        value_template: "{{ value_json.daily_import_from_grid | round(2) }}"
        last_reset_value_template: "{{ value_json.last_reset }}"
````

Should I remove the `last_reset_value_template` which seems to not be compatible with the state class ?

Thanks in advance for your help

michael-robbins, 2024-12-06:
There were some other issue threads about making it `total_increasing` => `total`

Ducatel, 2024-12-06:
By just removing the field `last_reset_value_template` it seems to work.

[image]

So I don't really know the purpose of this field. Do you know ?

bohdan-s, 2024-12-06:
It WAS required by HA, but they changed it in a newer version and I haven’t removed it yet….

Ducatel, 2024-12-06:
Ok no problem I can make a Pr for that

closed 2026-10-24 by bohdan-s