bohdan-s/SunGather #213: Docker hub images out of date

justin8, 2025-01-16:
When checking for fixes to some issues I saw, I found that the `bohdans/sungather` dockerhub image hasn't been updated in 10 months, but is still the recommended place to pull the image from in the readme.

Could we get it configured to auto-update on docker hub on new releases? I'd be happy to contribute a github action workflow for it if you want since dockerhub conveniently hides auto-builds behind a team/pro subscription these days.

bohdan-s, 2025-01-16:
Hi. I had set up a workflow to auto publish. It seems it has stoped working. 
ill have to have a look at what broke.

justin8, 2025-01-16:
Ah nice! Thanks.

justin8, 2025-01-16:
It looks like the current job only runs on tagged releases. but it's also only pushing the :latest tag and not the tagged version either - so the latest tagged version is 0.5.0 from 2 years ago. 

The last tagged commit is 0.5.3 which is seemingly the one on dockerhub from 10 months ago though. So I suspect the build still works, might just need a new release tag or to be fixed up to push latest on updates to main rather than tagged releases.