bohdan-s/SunGather #214: Support for SH15T

NoUserName32, 2025-01-19:
Can you please add support for SH15T (device_type_code 3621)

hardrock63, 2025-05-26:
I've the same problem.
I have used al the connection: sungrow, modbus, http bau same errors
This is my log
025-05-26 14:25:30 INFO     Serial detection failed, please set serial number in config.py
2025-05-26 14:25:30 INFO     Closing Session: ModbusTcpClient(192.168.1.90:502)
2025-05-26 14:25:30 INFO     Loading Export: exports\webserver
2025-05-26 14:25:30 INFO     Webserver: Configured
2025-05-26 14:25:30 INFO     Loading Export: exports\mqtt
2025-05-26 14:25:30 ERROR    Failed loading export: 'NoneType' object has no attribute 'replace'
			     Please make sure mqtt.py exists in the exports folder
Failed loading export: 'NoneType' object has no attribute 'replace'
			     Please make sure mqtt.py exists in the exports folder
2025-05-26 14:25:30 INFO     Modbus, Connecting new session
2025-05-26 14:25:30 ERROR    Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
2025-05-26 14:25:30 ERROR    Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
2025-05-26 14:25:30 WARNING  No data returned for read, 5000:38
No data returned for read, 5000:38
2025-05-26 14:25:30 ERROR    Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
2025-05-26 14:25:30 WARNING  No data returned for read, 5039:61
No data returned for read, 5039:61
2025-05-26 14:25:30 ERROR    Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
2025-05-26 14:25:30 WARNING  No data returned for read, 5100:100
No data returned for read, 5100:100
2025-05-26 14:25:30 ERROR    Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
2025-05-26 14:25:30 WARNING  No data returned for hold, 4999:10
No data returned for hold, 4999:10
2025-05-26 14:25:30 INFO     Disconnecting: ModbusTcpClient(192.168.1.90:502)
2025-05-26 14:25:30 INFO     Disconnecting: None
2025-05-26 14:25:30 WARNING  Data collection failed, skipped exporting data. Retying in 30 secs
Data collection failed, skipped exporting data. Retying in 30 secs
2025-05-26 14:25:30 INFO     Next scrape in 29 secs
2025-05-26 14:26:00 INFO     Modbus client is not connected, attempting to reconnect
2025-05-26 14:26:00 INFO     Connection: ModbusTcpClient(192.168.1.90:502)
2025-05-26 14:26:00 ERROR    Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
2025-05-26 14:26:03 ERROR    Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
2025-05-26 14:26:03 WARNING  No data returned for read, 5000:38
No data returned for read, 5000:38
2025-05-26 14:26:03 ERROR    Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
2025-05-26 14:26:03 WARNING  No data returned for read, 5039:61
No data returned for read, 5039:61
2025-05-26 14:26:03 ERROR    Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
2025-05-26 14:26:03 WARNING  No data returned for read, 5100:100
No data returned for read, 5100:100
2025-05-26 14:26:03 ERROR    Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.90, 502) failed: [Errno 111] Connection refused
2025-05-26 14:26:03 WARNING  No data returned for hold, 4999:10
No data returned for hold, 4999:10
2025-05-26 14:26:03 INFO     Disconnecting: ModbusTcpClient(192.168.1.90:502)
2025-05-26 14:26:03 INFO     Disconnecting: None
2025-05-26 14:26:03 WARNING  Data collection failed, skipped exporting data. Retying in 30 secs
Data collection failed, skipped exporting data. Retying in 30 secs
2025-05-26 14:26:03 INFO     Next scrape in 26 secs

Freakapotamus, 2025-11-28:
I am having issues also connecting to my SH15T.

This is my log:
2025-11-28 13:46:07 INFO     Starting SunGather 0.5.2
2025-11-28 13:46:07 INFO     Need Help? https://github.com/bohdan-s/SunGather
2025-11-28 13:46:07 INFO     NEW HomeAssistant Add-on: https://github.com/bohdan-s/hassio-repository
2025-11-28 13:46:07 INFO     Loaded config: config.yaml
2025-11-28 13:46:07 INFO     Loaded registers: registers-sungrow.yaml
2025-11-28 13:46:07 INFO     Registers file version: 0.2.4
2025-11-28 13:46:07 INFO     Logging to console set to: DEBUG
2025-11-28 13:46:07 DEBUG    Inverter Config Loaded: {'host': '192.168.1.43', 'port': 502, 'timeout': 10, 'retries': 3, 'slave': 1, 'scan_interval': 30, 'connection': 'modbus', 'model': None, 'smart_meter': False, 'use_local_time': False, 'log_console': 'DEBUG', 'log_file': 'OFF', 'level': 1}
2025-11-28 13:46:07 INFO     Loading SungrowClient 0.1.0
2025-11-28 13:46:07 DEBUG    Checking Modbus Connection
2025-11-28 13:46:07 INFO     Modbus client is not connected, attempting to reconnect
2025-11-28 13:46:07 INFO     Connection: ModbusTcpClient(192.168.1.43:502)
2025-11-28 13:46:07 DEBUG    Connection to Modbus server established. Socket ('192.168.1.4', 38129)
2025-11-28 13:46:10 DEBUG    load_registers: read, 4999:1
2025-11-28 13:46:10 DEBUG    Current transaction state - IDLE
2025-11-28 13:46:10 DEBUG    Running transaction 1
2025-11-28 13:46:10 DEBUG    SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x87 0x0 0x1
2025-11-28 13:46:10 DEBUG    New Transaction state 'SENDING'
2025-11-28 13:46:10 DEBUG    Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2025-11-28 13:46:10 DEBUG    Transaction failed. ([Errno 104] Connection reset by peer)
2025-11-28 13:46:10 DEBUG    Processing:
2025-11-28 13:46:10 DEBUG    Getting transaction 1
2025-11-28 13:46:10 DEBUG    Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2025-11-28 13:46:10 WARNING  Modbus connection failed
Modbus connection failed
2025-11-28 13:46:10 DEBUG    Modbus Error: [Input/Output] [Errno 104] Connection reset by peer
2025-11-28 13:46:10 INFO     Model detection failed, please set model in config.py
2025-11-28 13:46:10 DEBUG    load_registers: read, 4989:10
2025-11-28 13:46:10 DEBUG    Connection to Modbus server established. Socket ('192.168.1.4', 49179)
2025-11-28 13:46:10 DEBUG    Current transaction state - TRANSACTION_COMPLETE
2025-11-28 13:46:10 DEBUG    Running transaction 2
2025-11-28 13:46:10 DEBUG    SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x7d 0x0 0xa
2025-11-28 13:46:10 DEBUG    New Transaction state 'SENDING'
2025-11-28 13:46:10 DEBUG    Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2025-11-28 13:46:10 DEBUG    Transaction failed. ([Errno 104] Connection reset by peer)
2025-11-28 13:46:10 DEBUG    Processing:
2025-11-28 13:46:10 DEBUG    Getting transaction 2
2025-11-28 13:46:10 DEBUG    Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2025-11-28 13:46:10 WARNING  Modbus connection failed
Modbus connection failed
2025-11-28 13:46:10 DEBUG    Modbus Error: [Input/Output] [Errno 104] Connection reset by peer
2025-11-28 13:46:10 INFO     Serial detection failed, please set serial number in config.py
2025-11-28 13:46:10 INFO     Closing Session: ModbusTcpClient(192.168.1.43:502)
2025-11-28 13:46:10 INFO     Loading Export: exports console
+----------------------------------------------+
| Inverter Configuration Settings              |
+----------------------------------------------+
| Config             | Value                   |
+--------------------+-------------------------+
| host               | 192.168.1.43            |
| port               | 502                     |
| timeout            | 10                      |
| retries            | 3                       |
| RetryOnEmpty       | False                   |
| model              | None                    |
| serial_number      | None                    |
| level              | 1                       |
| scan_interval      | 30                      |
| use_local_time     | False                   |
| smart_meter        | False                   |
| connection         | modbus                  |
| slave              | 1                       |
| start_time         |                         |
+----------------------------------------------+
2025-11-28 13:46:10 DEBUG    Checking Modbus Connection
2025-11-28 13:46:10 INFO     Modbus, Connecting new session
2025-11-28 13:46:10 DEBUG    Connection to Modbus server established. Socket ('192.168.1.4', 55871)
[... 25 lines cut ...]
2025-11-28 13:46:10 DEBUG    Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2025-11-28 13:46:10 WARNING  Modbus connection failed
Modbus connection failed
2025-11-28 13:46:10 DEBUG    Modbus Error: [Input/Output] [Errno 104] Connection reset by peer
2025-11-28 13:46:10 DEBUG    Scraping: hold, 4999:10
2025-11-28 13:46:10 DEBUG    load_registers: hold, 4999:10
2025-11-28 13:46:10 DEBUG    Connection to Modbus server established. Socket ('192.168.1.4', 46485)
2025-11-28 13:46:10 DEBUG    Current transaction state - TRANSACTION_COMPLETE
2025-11-28 13:46:10 DEBUG    Running transaction 5
2025-11-28 13:46:10 DEBUG    SEND: 0x0 0x5 0x0 0x0 0x0 0x6 0x1 0x3 0x13 0x87 0x0 0xa
2025-11-28 13:46:10 DEBUG    New Transaction state 'SENDING'
2025-11-28 13:46:10 DEBUG    Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
[... cut ...]
Modbus connection failed
2025-11-28 13:46:10 DEBUG    Modbus Error: [Input/Output] [Errno 104] Connection reset by peer
2025-11-28 13:46:10 INFO     Disconnecting: ModbusTcpClient(192.168.1.43:502)
2025-11-28 13:46:10 INFO     Disconnecting: None
2025-11-28 13:46:10 WARNING  Data collection failed, skipped exporting data. Retying in 30 secs
Data collection failed, skipped exporting data. Retying in 30 secs
2025-11-28 13:46:10 DEBUG    Processing Time: 0.36 secs
2025-11-28 13:46:10 INFO     Next scrape in 29 secs