bohdan-s/SunGather #216: How to avoid display of offline data?

monojk, 2025-02-16:
When the server is not reachable the browser (e.g. chrome) displays offline cached data, that means incorrect data. How can I avoid this on the server side? I don't want a browser side solution as it would avoid display offline data from any other server.

bohdan-s, 2025-02-16:
Sorry.  Not sure exactly what you are asking?

monojk, 2025-02-16:
It's about the webserver.py code (export_webserver class). In cases when there is no connection possible to the raspberry hosting the SunGather, my browser displays an offline page, i.e. cached version with older data. That old data might be misleading.

I guess that there could be some server side setting that marks cached data (from a previous connection) as not to be shown w/o active connection? 

The reference seems to be too old: https://stackoverflow.com/questions/7597472/how-to-turn-off-html5-offline-caching