bohdan-s/SunGather #218: Home assistant MQTT sensors duplication

MattiaPette, 2025-03-01:
Hi everyone,
I've just configured an instance of Sungather (with the docker container) to publish data to mqtt in order to integrate the data to my HA setup.

I've noticed that after every power cycle of my mini-server (scheduled at 4am everyday) the sensor entities are duplicated with a _2, _3, _4 at the end of the name instead of reusing the old ones and i need to reconfigure my dashboard every time.
Is there a solution to this?

kaypohl, 2025-03-05:
I don't have a solution, but the same problem. I am also interested, why that happens.

BR
Kay

bohdan-s, 2025-03-06:
Hi, the issue everyone Sungather starts it sends a special MQTT message to tell HA to set up the sensors. 
These are persistent in MQTT server, so the next time HA starts it loads them again and HA sees them and recreates new sensors. 
Restarting MQTT and Sungather would be one work around I can think of.

kaypohl, 2025-03-06:
I followed your suggestion. Restarting MQTT and Sungather doesn't work. Maybe I need to update Sungather to the newest version.