bohdan-s/SunGather #219: Connection refused

monojk, 2025-03-28:
I have been running Sungather for months now. Code hasen't changed on my side but starting 3 days ago, it cannot gather data anymore. Even adding the serial in config.yaml don't help.
```
pi@rp4b:~/SunGather/SunGather $ /usr/bin/python3 -u /home/pi/SunGather/SunGather/sungather.py -v 10 --runonce
2025-03-28 09:50:54 INFO     Starting SunGather 0.5.1
2025-03-28 09:50:54 INFO     Need Help? https://github.com/bohdan-s/SunGather
2025-03-28 09:50:54 INFO     NEW HomeAssistant Add-on: https://github.com/bohdan-s/hassio-repository
2025-03-28 09:50:54 INFO     Loaded config: config.yaml
2025-03-28 09:50:56 INFO     Loaded registers: registers-sungrow.yaml
2025-03-28 09:50:56 INFO     Registers file version: 0.2.3
2025-03-28 09:50:56 INFO     Logging to console set to: DEBUG
2025-03-28 09:50:56 DEBUG    Inverter Config Loaded: {'host': '192.168.178.124', 'port': 8082, 'timeout': 20, 'retries': 3, 'slave': 1, 'scan_interval': 60, 'connection': 'http', 'model': 'SH8.0RT', 'smart_meter': False, 'use_local_time': False, 'log_console': 'WARNING', 'log_file': 'OFF', 'level': 3}
2025-03-28 09:50:56 INFO     Loading SungrowClient 0.1.0
2025-03-28 09:50:56 DEBUG    Checking Modbus Connection
2025-03-28 09:50:56 INFO     Modbus client is not connected, attempting to reconnect
2025-03-28 09:50:56 INFO     Connection: SungrowModbusWebClient_0.3.2(192.168.178.124:8082)
2025-03-28 09:50:56 DEBUG    Connection to websocket server failed: ws://192.168.178.124:8082/ws/home/overview, Message: [Errno 111] Connection refused
2025-03-28 09:50:59 INFO     Bypassing Model Detection, Using config: SH8.0RT
2025-03-28 09:50:59 DEBUG    load_registers: read, 4989:10
2025-03-28 09:50:59 DEBUG    Connection to websocket server failed: ws://192.168.178.124:8082/ws/home/overview, Message: [Errno 111] Connection refused
2025-03-28 09:50:59 WARNING  No data returned for read, 4989:10
No data returned for read, 4989:10
2025-03-28 09:50:59 DEBUG    Modbus Error: [Connection] Failed to connect[SungrowModbusWebClient_0.3.2(192.168.178.124:8082)]')
2025-03-28 09:50:59 INFO     Serial detection failed, please set serial number in config.py
2025-03-28 09:50:59 INFO     Loading Export: exports\console
+----------------------------------------------+
| Inverter Configuration Settings              |
+----------------------------------------------+
| Config             | Value                   |
+--------------------+-------------------------+
| host               | 192.168.178.124         |
| port               | 8082                    |
| timeout            | 20                      |
| retries            | 3                       |
| RetryOnEmpty       | False                   |
| model              | SH8.0RT                 |
| serial_number      | None                    |
| level              | 3                       |
| scan_interval      | 60                      |
| use_local_time     | False                   |
| smart_meter        | False                   |
| connection         | http                    |
| slave              | 1                       |
| start_time         |                         |
+----------------------------------------------+
2025-03-28 09:50:59 INFO     Loading Export: exports\webserver
2025-03-28 09:50:59 ERROR    Webserver: Error: [Errno 98] Address already in use
Webserver: Error: [Errno 98] Address already in use
2025-03-28 09:50:59 DEBUG    Checking Modbus Connection
2025-03-28 09:50:59 INFO     Modbus, Connecting new session
2025-03-28 09:50:59 DEBUG    Connection to websocket server failed: ws://192.168.178.124:8082/ws/home/overview, Message: [Errno 111] Connection refused
2025-03-28 09:50:59 DEBUG    Scraping: read, 4949:50
2025-03-28 09:50:59 DEBUG    load_registers: read, 4949:50
2025-03-28 09:50:59 DEBUG    Connection to websocket server failed: ws://192.168.178.124:8082/ws/home/overview, Message: [Errno 111] Connection refused
2025-03-28 09:50:59 WARNING  No data returned for read, 4949:50
```

michbeck100, 2025-03-28:
Seems like your inverter is refusing the connection: 
`Connection to websocket server failed: ws://192.168.178.124:8082/ws/home/overview, Message: [Errno 111] Connection refused`. Maybe restart on the inverter needed?

monojk, 2025-03-28:
I can access the system via Sungrow's app. Is there a way to restart the inverter other than switching the PV, battery and mains off?

pohly, 2025-03-28:
There has been a firmware update this week which disabled port 8082. I had the same problem and solved it by switching the access method to `connection: modbus` via port 502.

One other change: I had to change the scan interval from 10 s to 30 s because sampling sometimes (usually?) ran for more than 20 seconds.

meisel2000, 2025-03-28:
Same for my sungrow inverter. Port 8082 is disabled now.

pohly, 2025-03-28:
Let me add that it not only got disabled, the support is gone completely. As admin one can control whether access through port 502 is enabled, but there is nothing for 8082.

Paulmx2, 2025-03-29:
I also have the same problem.  My sh10rt had to be restarted 3 days ago and the "http" connection stopped providing data to Home Assistant via Sungather.  A switch to the Modbus connection in Sungather configuration does partially work but some registers don't work such as register 13023  battery_level always states 100.0 %.  
I also get  lots of errors in the log  "WARNING  Modbus connection failed".   
Is this definitely a Sungrow update that caused this issue?

michbeck100, 2025-03-29:
I guess your modbus is not working yet. I am using modbus on my SH10RT since the beginning. Never had big issues. And sungather didn’t change in the past. So it’s probably on your side.

Paulmx2, 2025-03-29:
I had Modbus problems when I first installed the system 18 months ago but http just worked so I stuck with that.  
Seems modbus is the preferred method but does anyone what actually happened to "http"?
Where can one see the inverter update history?

pohly, 2025-03-29:
I got an email from system@isolarcloud.com with the request to enable auto-updates because a new firmware version would be needed to "secure" my inverters (I have two). One got updated first, so I could verify that the web interface was different. I've not found the exact firmware version yet.

You can log in with user/pw1111 oder admin/pw8888, unless someone changed the passwords.

monojk, 2025-03-29:
How would I login with modbus port 502?

pohly, 2025-03-29:
You can access a local web interface via `https://<ip address or host name of your inverter>`.

meisel2000, 2025-03-29:
Maybe the firmware changes are related to
https://www.[forescout.com/resources/sun-down-research-report/](https://www.forescout.com/resources/sun-down-research-report/)
and
https://en.sungrowpower.com/security-notice-detail-2/5961

puijken, 2025-03-30:
Can confirm, same behaviour. Suddenly no more reporting and interface changed a little bit and is now HTTPS. Ater enabling (or rather it was enabled by default) Modbus and switching the Sungather config file to:

_connection: modbus
port: 502_

Things started to run fine again. Thanks a lot for pre-solving this.

closed 2025-04-19 by monojk