bohdan-s/SunGather #223: Sungrow SH6.0RT after firmware backup: only Port 443 is now open and Sungather fails when using SungrowModbusWebClient

AFI4711, 2025-04-25:
I use this excellent software Sungather since several years successfully. Now, since the last (and only?) update of the firmware of both the inverter and the communication modul WiNet-S port 80 and 8080 seems to be not open any longer.

**The logfile:**

```
sungather  | 2025-04-25 15:28:45 INFO     Starting SunGather 0.5.2
sungather  | 2025-04-25 15:28:45 INFO     Need Help? https://github.com/bohdan-s/SunGather
sungather  | 2025-04-25 15:28:45 INFO     NEW HomeAssistant Add-on: https://github.com/bohdan-s/hassio-repository
sungather  | 2025-04-25 15:28:45 INFO     Loaded config: /config/config.yaml
sungather  | 2025-04-25 15:28:46 INFO     Loaded registers: registers-sungrow.yaml
sungather  | 2025-04-25 15:28:46 INFO     Registers file version: 0.2.4
sungather  | 2025-04-25 15:28:46 INFO     Logging to console set to: INFO
sungather  | 2025-04-25 15:28:46 INFO     Logging to file set to: INFO
sungather  | 2025-04-25 15:28:46 INFO     Loading SungrowClient 0.1.0
sungather  | 2025-04-25 15:28:46 INFO     Modbus client is not connected, attempting to reconnect
sungather  | 2025-04-25 15:28:46 INFO     Connection: SungrowModbusWebClient_0.3.2(192.168.0.135:8082)
sungather  | 2025-04-25 15:28:49 WARNING  No data returned for read, 4999:1
sungather  | No data returned for read, 4999:1
sungather  | 2025-04-25 15:28:49 INFO     Model detection failed, please set model in config.py
sungather  | 2025-04-25 15:28:49 WARNING  No data returned for read, 4989:10
sungather  | No data returned for read, 4989:10
sungather  | 2025-04-25 15:28:49 INFO     Serial detection failed, please set serial number in config.py
sungather  | 2025-04-25 15:28:49 INFO     Loading Export: exports\console
sungather  | 2025-04-25 15:28:50 INFO     Loading Export: exports\influxdb
sungather  | 2025-04-25 15:28:50 ERROR    InfluxDB: Configured to use daily_pv_generation but not configured to scrape this register
sungather  | InfluxDB: Configured to use daily_pv_generation but not configured to scrape this register
sungather  | 2025-04-25 15:28:50 INFO     Loading Export: exports\mqtt
sungather  | 2025-04-25 15:28:50 ERROR    Failed loading export: 'NoneType' object has no attribute 'replace'
sungather  |                         Please make sure mqtt.py exists in the exports folder
sungather  | Failed loading export: 'NoneType' object has no attribute 'replace'
sungather  |                         Please make sure mqtt.py exists in the exports folder
sungather  | 2025-04-25 15:28:50 INFO     Modbus, Connecting new session
sungather  | 2025-04-25 15:28:50 WARNING  No data returned for read, 5000:38
sungather  | No data returned for read, 5000:38
sungather  | 2025-04-25 15:28:50 WARNING  No data returned for read, 5039:61
sungather  | No data returned for read, 5039:61
sungather  | 2025-04-25 15:28:50 WARNING  No data returned for read, 5100:100
sungather  | No data returned for read, 5100:100
sungather  | 2025-04-25 15:28:50 WARNING  No data returned for hold, 4999:10
sungather  | No data returned for hold, 4999:10
sungather  | 2025-04-25 15:28:50 WARNING  No data returned for hold, 5009:10
sungather  | No data returned for hold, 5009:10
sungather  | 2025-04-25 15:28:50 WARNING  No data returned for hold, 5034:10
sungather  | No data returned for hold, 5034:10
sungather  | 2025-04-25 15:28:50 INFO     Disconnecting: SungrowModbusWebClient_0.3.2(192.168.0.135:8082)
sungather  | 2025-04-25 15:28:50 INFO     Disconnecting: None
sungather  | 2025-04-25 15:28:50 WARNING  Data collection failed, skipped exporting data. Retying in 300 secs
sungather  | Data collection failed, skipped exporting data. Retying in 300 secs
sungather  | 2025-04-25 15:28:50 INFO     Next scrape in 299 secs
```

**The config file:**

```
inverter:
  host: 192.168.0.135                       # [Required] IP Address of the Inverter or Dongle
  # port: 502                               # [Optional] Default for modbus is 502, for http is 8082
  # timeout: 10                             # [Optional] Default is 10, how long to wait for a connection
  # retries: 3                              # [Optional] Default is 3, how many times to retry if connection fails
  # slave: 0x01                             # [Optional] Default is 0x01
  scan_interval: 300                        # [Optional] Default is 30
  connection: http                          # [Required] options: modbus, sungrow, http
  # model: "SG7.0RT"                        # [Optional] This is autodetected on startup, only needed if detection issues or for testing
                                            # See model list here: https://github.com/bohdan-s/SunGather#supported
  smart_meter: True                       # [Optional] Default is False, Set to true if inverter supports reading grind / house consumption
  use_local_time: True
  #log_console: ERROR                       # [Optional] Default is WARNING, Options: DEBUG, INFO, WARNING, ERROR
  log_console: INFO                        # [Optional] Default is WARNING, Options: DEBUG, INFO, WARNING, ERROR
  log_file: INFO                         # [Optional] Default is OFF, Options: OFF, DEBUG, INFO, WARNING, ERROR
  level: 2                                # [Optional] Set the amount of information to gather
                                            # 0 = Model and Solar Generation, 
                                            # 1 (default) = Useful data, all required for exports, 
                                            # 2 everything your Inverter supports, 
                                            # 3 Everything from every register 

# If you do not want to use a export, you can either remove the whole configuration block
# or set enabled: False
exports:
  # Print Registers to console, good for debugging / troubleshooting
  - name: console         
    enabled: True                          # [Optional] Default is False

  # Output data to InfluxDB
  - name: influxdb
    enabled: True                         # [Optional] Default is False
    url: "http://192.168.0.14:8086"            # [Optional] Default URL is http://localhost:8086
    token: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
    # username:                             # [Optional] Username if not using token
    # password:                             # [Optional] Password if not using token

    org: "privat"                           # [Required] InfluxDB Organization (for influxdb v1.8x this will be ignored)
    bucket: "haus"                          # [Required] InfluxDB Bucket (for influxdb v1.8x this is the database name)
    measurements:                           # [Required] Registers to publish to bucket
      - point: "state"
        register: timestamp
      - point: "power"
        register: daily_pv_generation
      - point: "power"
        register: load_power_hybrid
      - point: "power"
        register: total_dc_power
      - point: "power"
        register: export_to_grid
      - point: "power"
        register: daily_pv_export
      - point: "power"
        register: daily_import_energy
      - point: "device"
        register: internal_temperature
      - point: "device"
        register: run_state
      - point: "battery"
        register: battery_temperature
      - point: "battery"
        register: battery_level
      - point: "battery"
        register: battery_state_of_healthy
      - point: "energy"
        register: daily_battery_discharge_energy
      - point: "energy"
        register: daily_battery_charge_from_pv
      - point: "energy"
        register: daily_direct_energy_consumption
      - point: "statistic"
        register: direct_power_consumption_monthly_pv
      - point: "statistic"
        register: export_power_from_pv_monthly
      - point: "statistic"
        register: battery_charge_power_from_pv_monthly

  # Publish Registers to MQTT / Home Assistant
  - name: mqtt
    enabled: True                           # [Optional] Default is False
    host: 192.168.0.14                      # [Required] IP or Hostname of MQTT Server 
    #host: localhost                      # [Required] IP or Hostname of MQTT Server 
    # port: 1883                            # [Optional] Default 1883
    # topic: "inverter/{model}/registers"   # [Optional] Default: "inverter/{model}/registers", Variable {model} will be replaced with model number of inverter
    username: USERID                 # [Optional] Username is MQTT server requires it
    password: XXXXXXXX                    # [Optional] Password is MQTT server requires it

```

**The ports:**

```
Nmap scan report for espressif (192.168.0.135)
Host is up (0.0070s latency).
Not shown: 999 closed tcp ports (conn-refused)
PORT    STATE SERVICE
443/tcp open  https
```

Does anyone have any good advice for me?
Thanks a lot in advance!

pohly, 2025-04-25:
Related: https://github.com/bohdan-s/SunGather/issues/219#issuecomment-2764449532

If modbus on port 503 does not work for you, log into the web interface of the inverter as admin/admin and enable it.

AFI4711, 2025-04-25:
Should I use the hotspot of the WiNet-S Modul (using admin and pw8888 hopefully) or http[s]://IP-Address (USER?/PW?)  ?

AFI4711, 2025-04-26:
In the web interface I can't find the part about ports

pohly, 2025-04-26:
I have it under "System/Forwarding Configuration" in the "MODBUS" tab. The URL is `https://<ip address or machine name>/#/system/agreements/ProtocolConfigure`.

pohly, 2025-04-26:
There's a switch for port 502 (not 503 as a I wrote earlier - I was quoting from memory) without SSL and port 516 with SSL.

AFI4711, 2025-04-26:
Thanks a lot. I found it. Unfortunately I'll looking for ports 80 and 8080 `(SungrowModbusWebClient_0.3.2(192.168.0.135:8082)`  respectively

AFI4711, 2025-04-26:
after switching OFF | ON Port 502 connection is now possible!

telnet 192.168.0.135 502
Trying 192.168.0.135...
Connected to 192.168.0.135.
Escape character is '^]'.

Strange, but it works. Unfortunately port 80/8080 are closed so far.

closed 2025-04-26 by AFI4711